chipsalliance/Cores-VeeR-EL2 #5: Question: ASIC synthesis flow

nagendragd, 2020-06-25:
I am looking at exploring an ASIC RV32 implementation for area/power studies.
Would you have a document that describes your ASIC synthesis flow? 
Since I am looking at academic work here, I have access to the open Yosys tool (for synthesis) and an open source 45nm library. 

Thanks for any help or pointers to information.

Best Regards
Nagendra

aprnath, 2020-06-26:
Hi Nagendra,
We do not have such a document that can be shared publicly, since such flows are usually heavily customized to technology, tools and vendors. You may be able to find some flows available from academia or other public sources and apply to SweRV.

Regards

nagendragd, 2020-06-26:
Yes that makes sense. I will cobble up an ASIC flow using some open source tools so that I can test the RV32 PPA. 

Thanks
Nagendra

closed 2020-06-26 by nagendragd