chipsalliance/Cores-VeeR-EL2 #6: [HDL 9-849]

altuSemi, 2020-08-18:
Hi,

Thanks for the repository.
I got this error in Vivado 2020.1:
[HDL 9-849] Syntax error : file ended before end of clause. ["../Cores-SweRV-EL2/snapshots/default/el2_param.vh":156]
Seems like the parameter instance is not parsed correctly.

Also, what is the highest frequency I can expect on FPGA?

Thanks

agrobman, 2020-08-18:
do you see compilation problems with other tools verilator/vcs/xrun?

FPGA implementation should run at 50-100MHz ...

olofk, 2020-08-18:
I haven't tried with Vivado 2020.1 but it builds fine in vivado 2019.1 when using FuseSoC. If you install FuseSoC, you can then add SweRV EL2 as a library with `fusesoc library add el2 https://github.com/chipsalliance/Cores-SweRV-EL2` and run 'fusesoc run --target=synth chipsalliance.org:cores:SweRV_EL2` to have it building for vivado.

olofk, 2020-08-18:
hmm.. I just noticed that #3 isn't applied yet, so the FuseSoC builds are currently broken

JavariaK, 2020-12-31:
> Eventually it was a config issue, closing.
> Thanks

Can you describe what was the issue and how was it resolved?
Thank you