chipsalliance/Cores-VeeR-EL2 #12: coremark simulation results

kidonglee, 2020-10-20:
Coremark score from my simulation is lower than official score (3.6).
My simulation results are as follows, 
    - default + cmark : 2.42
    - high_perf + cmark : 2.42
    - high_perf + cmark_dccm : 3.42
    - high_perf + cmark_iccm : 3.43
Please let me know how to get official coremark score (3.6).
Thank you.

aprnath, 2020-10-21:
Hi, we will get back to you with the compiler switches needed to achieve the published score. 
Thanks.

robertgolla, 2020-10-27:
So cmark scores are dependent on which version of the compiler you use and which flags.

Give the command below a try.  With gcc 9.3 this yielded CM 3.61.

Thanks.

run.int make -f $RV_ROOT/tools/Makefile verilator target=default CONF_PARAMS='-set=btb_size=512 -set=bht_size=2048' TEST=cmark_dccm TEST_CFLAGS='-finline-limit=400 -mbranch-cost=1 -Ofast -fno-code-hoisting -funroll-all-loops'
 
VerilatorTB: Start of sim
 
DCCM pre-load from f0040000 to f00416d0
2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 277305
Total time (secs): 277
Iterat/Sec/MHz   : 3.61
Iterations       : 1
Memory location  : STATIC
seedcrc          : 0xo9p5
[0]crclist       : 0xo714
[0]crcmatrix     : 0x1pn7
[0]crcstate      : 0x8o3k
[0]crcfinal      : 0xo714
Correct operation validated. See readme.txt for run and reporting rules.
TEST_PASSED

robertgolla, 2020-10-27:
Correction it was actually gcc 9.2 for this result.  Not sure if 9.3 would be different or not.

closed 2020-10-28 by aprnath