chipsalliance/Cores-VeeR-EL2 #13: cmark scores

kidonglee, 2020-10-30:
So cmark scores are dependent on which version of the compiler you use and which flags.

Give the command below a try.  With gcc 9.3 this yielded CM 3.61.

Thanks.

run.int make -f $RV_ROOT/tools/Makefile verilator target=default CONF_PARAMS='-set=btb_size=512 -set=bht_size=2048' TEST=cmark_dccm TEST_CFLAGS='-finline-limit=400 -mbranch-cost=1 -Ofast -fno-code-hoisting -funroll-all-loops'
 
VerilatorTB: Start of sim
 
DCCM pre-load from f0040000 to f00416d0
2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 277305
Total time (secs): 277
Iterat/Sec/MHz   : 3.61
Iterations       : 1
Memory location  : STATIC
seedcrc          : 0xo9p5
[0]crclist       : 0xo714
[0]crcmatrix     : 0x1pn7
[0]crcstate      : 0x8o3k
[0]crcfinal      : 0xo714
Correct operation validated. See readme.txt for run and reporting rules.
TEST_PASSED

_Originally posted by @robertgolla in https://github.com/chipsalliance/Cores-SweRV-EL2/issues/12#issuecomment-717528633_

aprnath, 2020-10-31:
Hi @kidonglee,
Sorry about that. That particular combination of compiler args increases the image size beyond the bounds of the tiny demo bench we have provided. Our internal EL2 demo bench has progressed since the last push to github. If you can wait a couple of weeks, we will be releasing some updates with bug fixes and enhancements to EL2, and it will have a better bench that will support larger programs, as well as the above combination of compiler options.

kidonglee, 2020-11-01:
> it will have

**Dear aprnath,**

Thanks for the kind explanation.
I am looking forward to experience the new one with  with bug fixes and enhancements.

Thanks again.

kidonglee, 2020-11-13:
> Hi @kidonglee,
> Sorry about that. That particular combination of compiler args increases the image size beyond the bounds of the tiny demo bench we have provided. Our internal EL2 demo bench has [...]

Dear aprnath,

Is there any progress on the error related to the cmark test with gcc9.2 ?
Please let me know if you get the solution.

Thanks.

aprnath, 2020-11-17:
Hi @kidonglee, please use the recently released EL2-1.3. It should give you the expected cmark with gcc 9.2.0.

closed 2020-11-17 by aprnath