chipsalliance/Cores-VeeR-EL2 #14: synthesis script template

kidonglee, 2020-11-02:
Hi,

I am trying to synthesis swerv-el2 by using design compiler.
I got some syntax error messages while analyzing it.
They seem to be related to 'package' and 'include'.
Actually, because I am not good at synthesis work, I don't know what to do.
Is there a kind of synthesis script template for swerv-el2 ?

Thanks in advance.

Codasip, 2020-11-02:
There is a "Pro" version of Core SweRV Support package (SSP), which contains tamplates and proven scripts for SweRV EL2 DC synthesis. You can use fusesoc flow with DC in a similar way as you have (probably) done with Vivado. Please contact sales@codasip.com for a help.

kidonglee, 2020-11-02:
Dear Codasip, 
Thanks for the advice.
Actually, I can't afford Pro version service yet.
I hope that I have a chance to use it someday.
I will try to find a way in fusesoc flow as you mentioned.
Thank you very much.

closed 2020-11-04 by kidonglee