chipsalliance/Cores-VeeR-EL2 #15: synthesis errors

kidonglee, 2020-11-04:
Hi, 

While synthesizing SweRV-EL2 by using dc(2016.03), I got some error messages (unknown width).
Those are related to pt parameter, as follows,
  1. $clog2(pt.xxx)
  2. (pt.xxx)'(sig)
And, I can fix the errors by changing the codes, as follows,
  1. localparam xxx=pt.xxx; $clog2(xxx);
  2. localparam xxx=pt.xxx; (pt)'(sig)
Please check what the problem is.

Thanks.

aprnath, 2020-11-06:
Hi @kidonglee, I checked with our PD guru and he concurred that the version of DC you use will choke on the param typecasting code. He suggested using a recent version such asc2017.09-SP4 or later, possible.

kidonglee, 2020-11-08:
Dear, aprnath.

Thank you for the reply.
The recent version of DC is not available for me right now.
I will give it  a try later.

Thanks, again.

closed 2020-11-08 by kidonglee