chipsalliance/Cores-VeeR-EL2 #16: Synthesis clock speed is very slow

kidonglee, 2020-11-09:
Hi, 

I synthesized SweRV-EL2 and Rocket Chip with default configuration
Rocket Chip can be synthesized up to 500MHz without timing slack.
However, SweRV-EL2 can be synthesized up to 300MHz without timing slack, in the same condition.
SweRV-EL2 is very slower than Rocket Chip.
Is it reasonable results?
Could you please give me  a comment on the result.

Thanks.

Codasip, 2020-11-09:
Can you please share your constraints you have used ?

kidonglee, 2020-11-10:
Dear Codasip,

The synthesis condition is as follow,
  - scale factor = 0.563  // margin
  - clk = 400MHz  // actual clock period = (1000/400)*0.563 = 1.41 ns
  - jtag_clk = 10MHz // actual clock period = (1000/10)*0.563 = 56.3 ns
  - clock transition value = 0.1

With above condition, the max timing slack is -0.24ns.
FYI, I post the longest delay path, as below.
Please check it.

Thanks

```
  Startpoint: swerv/ifu/aln/rdpff/dout_reg[1]
              (rising edge-triggered flip-flop clocked by clk)
  Endpoint: mem/iccm.iccm/red_lru/genblock.dffs/dout_reg[0]
            (rising edge-triggered flip-flop clocked by clk)
  Path Group: clk
  Path Type: max

  Point                                                   Incr       Path
  --------------------------------------------------------------------------
  clock clk (rise edge)                                   0.00       0.00
  clock network delay (ideal)                             0.00       0.00
  swerv/ifu/aln/rdpff/dout_reg[1]/CK                      0.00 #     0.00 r
  swerv/ifu/aln/rdpff/dout_reg[1]/Q                       0.12       0.12 r
  swerv/ifu/aln/U100/Y                                    0.02       0.15 r
  swerv/ifu/aln/U822/Y                                    0.02       0.17 f
  swerv/ifu/aln/U55/Y                                     0.02       0.19 r
  swerv/ifu/aln/U104/Y                                    0.02       0.21 f
  swerv/ifu/aln/U760/Y                                    0.02       0.22 r
  swerv/ifu/aln/U82/Y                                     0.01       0.24 f
  swerv/ifu/aln/U230/Y                                    0.01       0.25 r
  swerv/ifu/aln/U171/Y                                    0.02       0.28 r
  swerv/ifu/aln/U32/Y                                     0.02       0.29 f
  swerv/ifu/aln/U31/Y                                     0.02       0.31 r
  swerv/ifu/aln/U223/Y                                    0.02       0.33 f
  swerv/ifu/aln/U17/Y                                     0.07       0.40 r
  swerv/ifu/aln/U211/Y                                    0.03       0.44 f
  swerv/ifu/aln/U291/Y                                    0.02       0.46 r
  swerv/ifu/aln/U680/Y                                    0.02       0.48 f
  swerv/dec/instbuff/U59/Y                                0.02       0.50 f
  swerv/dec/decode/U1519/Y                                0.02       0.52 r
  swerv/dec/decode/U873/Y                                 0.02       0.54 f
  swerv/dec/decode/U1023/Y                                0.02       0.56 r
  swerv/dec/decode/U1008/Y                                0.02       0.58 f
  swerv/dec/decode/U181/Y                                 0.02       0.60 r
  swerv/dec/decode/U397/Y                                 0.02       0.62 f
  swerv/dec/decode/U182/Y                                 0.02       0.64 r
  swerv/dec/decode/U408/Y                                 0.02       0.66 f
  swerv/dec/decode/U71/Y                                  0.03       0.69 f
  swerv/dec/decode/U248/Y                                 0.02       0.71 r
  swerv/dec/decode/U221/Y                                 0.01       0.72 f
  swerv/dec/decode/U433/Y                                 0.02       0.74 r
  swerv/dec/decode/U419/Y                                 0.02       0.75 f
  swerv/exu/U448/Y                                        0.02       0.77 r
  swerv/exu/U222/Y                                        0.03       0.80 f
  swerv/exu/i_alu/U858/Y                                  0.03       0.83 f
  swerv/exu/i_alu/U54/Y                                   0.02       0.85 r
  swerv/exu/i_alu/U208/CON                                0.02       0.87 f
  swerv/exu/i_alu/U783/Y                                  0.02       0.89 r
  swerv/exu/i_alu/U320/Y                                  0.02       0.91 f
  swerv/exu/i_alu/U847/Y                                  0.02       0.93 r
  swerv/exu/i_alu/U794/Y                                  0.02       0.96 f
  swerv/exu/i_alu/U793/Y                                  0.02       0.98 f
  swerv/exu/i_alu/U127/Y                                  0.02       1.00 r
  swerv/exu/i_alu/U78/Y                                   0.02       1.01 f
  swerv/exu/i_alu/U618/Y                                  0.02       1.03 r
[... 25 lines cut ...]
  swerv/ifu/mem_ctl/U2031/Y                               0.02       1.45 r
  mem/iccm.iccm/U215/Y                                    0.03       1.48 r
  mem/iccm.iccm/U200/Y                                    0.04       1.52 r
  mem/iccm.iccm/U726/Y                                    0.02       1.54 f
  mem/iccm.iccm/U727/Y                                    0.02       1.56 r
  mem/iccm.iccm/U83/Y                                     0.02       1.58 f
  mem/iccm.iccm/U564/Y                                    0.02       1.60 r
  mem/iccm.iccm/U302/Y                                    0.02       1.62 f
  mem/iccm.iccm/U662/Y                                    0.01       1.64 r
  mem/iccm.iccm/red_lru/U3/Y                              0.02       1.66 r
  mem/iccm.iccm/red_lru/genblock.dffs/dout_reg[0]/D       0.00       1.66 r
  data arrival time                                                  1.66

  clock clk (rise edge)                                   1.41       1.41
  clock network delay (ideal)                             0.00       1.41
  mem/iccm.iccm/red_lru/genblock.dffs/dout_reg[0]/CK      0.00     1.41 r
  library setup time                                      0.01       1.42
  data required time                                                 1.42
  --------------------------------------------------------------------------
  data required time                                                 1.42
  data arrival time                                                 -1.66
  --------------------------------------------------------------------------
  slack (VIOLATED)                                                  -0.24

```

kidonglee, 2020-11-11:
Dear Codasip, 

Did you check my comment above ?
Actually, I wonder if the difference in speed between rocket and SweRV-EL2 is reasonable or not.
If you have compared the speed of them, could you please let me know the results ?
I am sorry to bother you but please help.

Thanks

aprnath, 2020-11-12:
Hi @kidonglee,
We are meeting ~600Mhz for 16nm technology, **worst case corner**.

Your scale factor seems very large! Based on that, your target frequency > 700Mhz. With the slack you show, the result is closer to 600. Why do you need such a large scale factor, BTW?

aprnath, 2020-11-12:
Do note that one of the reasons we do not provide synthesis scripts is that there are several key factors, such as technology node, library vendor, memory vendor/type, synthesis methodology, margins, clk skew, configuration, that will have a sizable impact on synthesis results.

kidonglee, 2020-11-12:
> Hi @kidonglee,
> We are meeting ~600Mhz for 16nm technology, **worst case corner**.
> 
> Your scale factor seems very large! Based on that, your target frequency > 700Mhz. With the slack you show, the result is closer to 600. Why do you need such a large scale factor, BTW?

Dear aprnath, 

I understand what you mean.
However, what I want to know is the difference between ocket and SweRV-EL2.
I just wonder if the difference of clock speed between them is reasonable or not.
If you also compared them, please let me know the results.

Than you very much for the kindness.