chipsalliance/Cores-VeeR-EL2 #21: How is store after faulting load handled?

Silabs-ArjanB, 2021-02-02:
Hello,

I see in the Reference Manual that a 'core load' (e.g. from DCCM) that has a fault will lead to a precise exception (table 2.4). Suppose the load fault is due to a failing ECC check (normally a late arriving signal) and suppose the faulting load is followed by a 'core store' (e.g. to the DCCM). Assuming that this store will be cancelled/suppressed, can you tell me which signal in the RTL takes care of that? Should I therefore also expect this signal to become a timing critical input to the DCCM?

deepakpan, 2021-02-02:
You can look at the design file "lsu_lsc_ctl.sv" inside the lsu. The double bit ECC error (lsu_double_bit_error_m/lsu_double_bit_error_r) will be indicated in the lsu_error_pkt_r to the TLU unit in R stage. TLU will flush the load(dec_tlu_i0_kill_writeb_r) and if there is a store behind it will get flushed as well.

Silabs-ArjanB, 2021-02-02:
Thank you very much for your answer. I will study those signals better first.

closed 2021-02-03 by Silabs-ArjanB