chipsalliance/Cores-VeeR-EL2 #23: Testbenches don't work for typical_pd using verilator

kuoyaoming93, 2021-04-01:
Hello,

I was trying to run tests for typical_pd target, but the program hangs at the beginning.
For example, when I run the test **dhry** (testbench/tests/dhry) using the following command:

`make TEST=dhry target=typical_pd`

The output of **exec.log** is: 
```
//   Cycle : #inst    0    pc    opcode    reg=value   ; mnemonic
        16 :       #1 0 80000000 595552b7  t0=59555000 ; lui t0, 0x59555000
        22 :       #2 0 80000004 55528293  t0=59555555 ; addi t0, t0, 1365
        28 :       #3 0 80000008 7c029073              ; csrrw zero, csr_7c0, t0
        33 :       #4 0 8000000c 70048117  sp=f004800c ; auipc sp, 0x70048000
        38 :       #5 0 80000010 63410113  sp=f0048640 ; addi sp, sp, 1588
        39 :       #6 0 80000014 5ee000ef  ra=80000018 ; jal ra, 0x80000602
        48 :       #7 0 80000602 f0040737  a4=f0040000 ; lui a4, 0xf0040000
        49 :       #8 0 80000606 0000716d  sp=f0048530 ; c.addi16sp -272
        50 :       #9 0 80000608 5b870713  a4=f00405b8 ; addi a4, a4, 1464
        51 :      #10 0 8000060c f00407b7  a5=f0040000 ; lui a5, 0xf0040000
        52 :      #11 0 80000610 5d878793  a5=f00405d8 ; addi a5, a5, 1496
        53 :      #12 0 80000614 00072f03  t5=00000000 ; lw t5, 0(a4) [f00405b8]
        54 :      #13 0 80000618 00472e83  t4=00000000 ; lw t4, 4(a4) [f00405bc]
        55 :      #14 0 8000061c 00872e03  t3=00000000 ; lw t3, 8(a4) [f00405c0]
        56 :      #15 0 80000620 0000d9da              ; c.swsp s6, 0xf0 [f0048620]
        63 :      #16 0 00000000 00000000              ; .short 0
        72 :      #17 0 00000000 00000000              ; .short 0
        76 :      #18 0 00000000 00000000              ; .short 0
        80 :      #19 0 00000000 00000000              ; .short 0
        84 :      #20 0 00000000 00000000              ; .short 0
        88 :      #21 0 00000000 00000000              ; .short 0
        92 :      #22 0 00000000 00000000              ; .short 0
        96 :      #23 0 00000000 00000000              ; .short 0
       100 :      #24 0 00000000 00000000              ; .short 0 
```

As we can see the program hangs at the beginning. I think it's related with the stack or the linker?
Is there an example with typical_pd that I could follow?

agrobman, 2021-04-01:
Seems the test/CPU tries to access above DCCM end address and the  CPU takes exception. the test code does not set up exceptions/CPU CSRs nor the test code has exception handlers, So the CPU constantly enters exceptions ...

I'm not sure what did you change in that target configuration, but the original typical_pd target has last DCCM address of:

`define RV_DCCM_EADR 32'hf0043fff

and the linker sets SP to  
 `      38 :       #5 0 80000010 e2410113  sp=f0043e30 ; addi    sp,sp,-476`
in our simulations

check your dhry.map to find why STACK var was set to so high address. 

( you also can increase DCCM size, or try to run cmark or cmark_dccm tests)

kuoyaoming93, 2021-04-01:
Thank you very much @agrobman for yout quick reply. 

I could run succesfully the cmark and cmark_dccm test using typical_pd configuration. I didn't change the original target configuration, just cloned the repo and run the tests.

The Linkerscript generated by **swerv.config**:
```
SECTIONS
{
  . = 0x80000000;
  .text.init .  : { *(.text.init) }
  
  .text . : { *(.text) }
  _end = .;
  . = 0xd0580000;
  .data.io .  : { *(.data.io) }
    . = 0xf0040000 ;
  .data  :  ALIGN(0x800) { *(.*data) *(.rodata*) STACK = ALIGN(16) + 0x8000; }
  .bss : { *(.bss) }
  
    . = 0xfffffff8; .data.ctl : { LONG(0xf0040000); LONG(STACK) }
}
```
Like you said the STACK is so high. I was watching cmark_dccm and the only diference is in this line:

`.data  :  ALIGN(0x800) { *(.*data) *(.rodata*) STACK = ALIGN(16) + 0x1000; }`

I copied the content of cmark_dccm.ld in the dhry folder and I could solve the problem. 

It seems that the STACK var is harcoded in **swerv.config** and it doesn't change for different target configurations (see line 2598 in configs/swerv.config).

agrobman, 2021-04-01:
end of swerv.config script should be like:

```
SECTIONS
{
  . = $config{reset_vec};
  .text.init .  : { *(.text.init) }
  $iccm
  .text . : { *(.text) }  
  _end = .;
  . = $io;
  .data.io .  : { *(.data.io) }
  $data_loc
  .data  :  ALIGN(0x800) { *(.*data) *(.rodata*)}
  .bss :  {BSS_START = .; *(.*bss)}
  BSS_END = .;
  STACK = ALIGN(16) + 0x1000;
  $iccm_ctl
  $dccm_ctl
}

EOF
    close FILE;
} #}}}
```

closed 2021-04-02 by kuoyaoming93