chipsalliance/Cores-VeeR-EL2 #24: Debug support

olofk, 2021-04-09:
I'm adding support for SweRV EL2 in SweRVolf. It seems to run fine except for the debug interface. I can read and write RAM but after trying to access device memory through the debugger there's an error messsage about failed abstract commands and then I'm losing connection. This might very well be something that is not correctly hooked up on my side, but I'm wondering about the state of the debug support in EL2. We recently made some changes in how OpenOCD talks to SwerRVolf with EH1 1.8+. Are these changes aligned in the latest EL2 version as well or do I need special handling?

agrobman, 2021-04-09:
EL2 should be compatible with latest EH1 in debugger logic. I.e. abstract memory commands to external addresses go via system bus port. Check your SB connections, check that you can access your peripherals with access size the openocd uses in abstract memory commands. (Ex can uart be accessed with 32 bit access?)
Is the RAM you read and write DCCM or DRAM? What method openocd is using to access the 'RAM'?

olofk, 2021-04-27:
Thanks for the info. I think I just had it wired up wrong. Closing for now since I don't there is anything actionable here

closed 2021-04-27 by olofk