chipsalliance/Cores-VeeR-EL2 #28: Pre-load Task in testbench

HamzaShabbir517, 2021-05-31:
As the Swerv have an method to preload the ICCM and DCCM when simulating it on verilator. Will the same task will work on ASIC flow or have to came up with other technology to pre-load the ICCM and DCCM

agrobman, 2021-05-31:
Depends what do u call "ASIC" flow ...

HamzaShabbir517, 2021-05-31:
Mean on actual hardware. Like FPGA board.

agrobman, 2021-05-31:
These are verilog modeling tasks . They can't be used with real silicone, like FPGA. You need to develop CCM preload drivers, using natural  data paths like DMA port or debugger memory access. ( DCCM can be preloaded by CPU code).

hello_world_iccm test shows how to preload ICCM with the CPU code, using HW LSU-DMA bridge. ASIC interconnect should have data path from CPU LSU master port to the slave DMA port. The demo TB has such simplified "interconnect" module to support this type of operation, demonstrated by this test.

HamzaShabbir517, 2021-05-31:
ok thankg you for ur response it help alot

HamzaShabbir517, 2021-05-31:
ok thanks for your quick response. It really helps me alot

closed 2021-05-31 by HamzaShabbir517