chipsalliance/Cores-VeeR-EL2 #29: pd_defines.vh

HamzaShabbir517, 2021-05-31:
While i was simulating the SweRV core on verilator. I found that it created few header files in the snapshot/default/ folder. One of the file is pd_define.vh. In that file I have encounter a line which says

`define TEC_RV_ICG HDBLVT16_CKGTPLT_V5_12

Now is the value after TEC_RV_ICG indicates some physical library cell which is used for clock gating. If yes than what does this mean as to which physical library it belongs. Does it belong to 65nm technology or something else and of which vendor?

Thanking you in advance

agrobman, 2021-05-31:
as cell name suggests it's 16nm technology  - you need to consult with your tech library vendor about particular cell names.
You need to consult also with your synthesis tool vendor how to enforce usage of specific cells/memories in final netlist.
Any guidance how to synthesize  SWERV cores is not provided here. These questions are out of scope / support of the DB

HamzaShabbir517, 2021-05-31:
ok thanks for the response

closed 2021-05-31 by HamzaShabbir517