chipsalliance/Cores-VeeR-EL2 #31: Synthesis error using Vivado 2017.4

asadaleem-lm, 2021-08-02:
I am working on latest repository and running the core through Vivado. On the synthesis step, syntax error is displayed on the terminal. On the further digging into the design , I found that [rvjtag](https://github.com/chipsalliance/Cores-SweRV-EL2/blob/master/design/dmi/rvjtag_tap.v) contains the systemVerilog constructs i.e. logic and always_comb while the file type is verilog. Am I getting it right or there is different reason for that?

agrobman, 2021-08-02:
u should have a way to tell Vivado that the whole design is written in SystemVerilog, regardless file name extensions.
(for ex Cadence xrun has  -sv switch or VCS has -sverilog switch)  Other users were able to build FPGA image with Vivado successfully

asadaleem-lm, 2021-08-03:
There are numerous workarounds for this issue. However, my concern is that if a file is using System Verilog constructs then its extension should be **.sv** not **.v**

closed 2021-08-03 by asadaleem-lm

olofk, 2021-08-10:
FYI, the FuseSoC core description file for SweRV EL2 sets the file type to System Verilog so if you use that, you would get the correct file type regardless of the file extension