chipsalliance/Cores-VeeR-EL2 #32: Synthesising the EL2 for a Intel Cyclone 10 GX FPGA

davidp135, 2021-08-16:
Hi,
I am trying to build the EL2 design for the Intel Cyclone 10 GX FPGA. This is using the free license within Quartus Prime Pro 21.2.

I am having issues with the `el2_param.vh `and `el2_pdef.vh` header files with the following error message:
`Error(13406): Verilog HDL error at el2_param.vh(1): object "el2_param_t" is not declared `

It looks like when Quartus evaluates el2_param.vh it does not know anything about the typedef in el2_pdef.vh.

I am including the header files in the design using the following command:
`set_global_assignment -name SEARCH_PATH "../../../tools/snapshots/default"`

I have attached the tcl script files I am using to build the project. They were located in a folder at CORES-SWERV-EL2/syn/cyclone10gx To build the project I run `quartus_sh -t test_app.tcl` in Linux. The .qpf project file will then be in the work folder below.

From what I understand the SystemVerilog support in Quartus Prime Pro is much better than in the older Lite software versions, so hopefully I am not too far off. Could anyone give me some advise on how to build the SweRV EL2 for this FPGA?

Any help would be much appreciated.

Thanks

David

[cyclone10gx.zip](https://github.com/chipsalliance/Cores-SweRV-EL2/files/6992821/cyclone10gx.zip)

agrobman, 2021-08-16:
Swerv.config script creates a few files in <work_dir>/snapshots/<snap> directory, these files need to be added to the compilation file list for synthesis, this directory should be also added to the search path of your tool.

generated files:

common_defines.vh or pd_defines.vh
el2_pdef.vh

should be compiled first they provide some verilog defines and the core parameter definition

( see tools/Makefile what is compiled for simulations,iust  ignore testbench files)

davidp135, 2021-08-16:
Hi agrobman, thanks for the response. I tried what I think you suggested and added the above 3 .vh files to synthesize before including the other header files and unfortunately I get the same error.

Here is my test_source.tcl file now:

```set_global_assignment -name SYSTEMVERILOG_FILE ../../../tools/snapshots/default/common_defines.vh
set_global_assignment -name SYSTEMVERILOG_FILE ../../../tools/snapshots/default/pd_defines.vh
set_global_assignment -name SYSTEMVERILOG_FILE ../../../design/include/el2_def.sv
set_global_assignment -name SYSTEMVERILOG_FILE ../../../tools/snapshots/default/el2_pdef.vh

set_global_assignment -name SEARCH_PATH "../../../tools/snapshots/default"

set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_swerv_wrapper.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_mem.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_pic_ctrl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_swerv.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_dma_ctrl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_aln_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_compress_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_ifc_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_bp_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_ic_mem.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_mem_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_iccm_mem.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_decode_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_gpr_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_ib_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_tlu_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_trigger.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/exu/el2_exu_alu_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/exu/el2_exu_mul_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/exu/el2_exu_div_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/exu/el2_exu.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_clkdomain.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_addrcheck.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_lsc_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_stbuf.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_bus_buffer.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_bus_intf.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_ecc.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_dccm_mem.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_dccm_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_trigger.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dbg/el2_dbg.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dmi/dmi_wrapper.v
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dmi/dmi_jtag_to_core_sync.v
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dmi/rvjtag_tap.v
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/el2_lib.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/beh_lib.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/mem_lib.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/ahb_to_axi4.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/axi4_to_ahb.sv

agrobman, 2021-08-16:
why is *tools/snapshots/default/* path to these files? 
Are you running make in the tools directory?

Normally make should be run in a work directory outside DB ...  It creates snapshots/default in the **work** directory ...
(BTW, did you run simulation of hello world test at least to check that your DB is OK?)

davidp135, 2021-08-17:
Good point with the _tools/snapshots/default/_ path, I have now built a snapshot in a separate work folder. The hello world test works fine with the config set  as `$RV_ROOT/configs/swerv.config -target=default"

I still do get the same error as above, but interestingly the synthesiser must be getting further before this error occurs. 

Firstly I get an error in `beh_lib.sv` because it doesn't like the ``module `TEC_RV_ICG`` name:
`Error(13411): Verilog HDL syntax error at beh_lib.sv(753) near text (', expecting '<identifier> `

Since I am doing an FPGA build I don't need this module so can comment it out. Then I am back to:
`Error(13406): Verilog HDL error at el2_param.vh(1): object "el2_param_t" is not declared `

Here is the source tcl script as it stands now
```
set_global_assignment -name SYSTEMVERILOG_FILE ../../../work/snapshots/default/common_defines.vh
set_global_assignment -name SYSTEMVERILOG_FILE ../../../work/snapshots/default/pd_defines.vh
set_global_assignment -name SYSTEMVERILOG_FILE ../../../design/include/el2_def.sv
set_global_assignment -name SYSTEMVERILOG_FILE ../../../work/snapshots/default/el2_pdef.vh

set_global_assignment -name SEARCH_PATH "../../../design/include"
set_global_assignment -name SEARCH_PATH "../../../work/snapshots/default"

set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_swerv_wrapper.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_mem.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_pic_ctrl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_swerv.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_dma_ctrl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_aln_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_compress_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_ifc_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_bp_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_ic_mem.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_mem_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu_iccm_mem.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/ifu/el2_ifu.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_decode_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_gpr_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_ib_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_tlu_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec_trigger.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dec/el2_dec.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/exu/el2_exu_alu_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/exu/el2_exu_mul_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/exu/el2_exu_div_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/exu/el2_exu.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_clkdomain.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_addrcheck.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_lsc_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_stbuf.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_bus_buffer.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_bus_intf.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_ecc.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_dccm_mem.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_dccm_ctl.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lsu/el2_lsu_trigger.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dbg/el2_dbg.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dmi/dmi_wrapper.v
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dmi/dmi_jtag_to_core_sync.v
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/dmi/rvjtag_tap.v
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/el2_lib.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/beh_lib.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/mem_lib.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/ahb_to_axi4.sv
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/lib/axi4_to_ahb.sv
```
I have attached the project also if that helps. It is all straight from master apart from the work and syn folders.

Thanks

David
[Cores-SweRV-EL2.zip](https://github.com/chipsalliance/Cores-SweRV-EL2/files/7001222/Cores-SweRV-EL2.zip)

agrobman, 2021-08-17:
David, make sure you have RV_FPGA_OPTIMIZE define in the common_defines.vh 
(-fpga_optimize switch of swerv.config removes clocks gating cells to increase CPU frequency for FPGA implementations in expense of increased power,)

BTW you don't need ahb_to axi4 and axi4_to_ahb modules if you build CPU with AXI busses

agrobman, 2021-08-17:
+ you need to have your <work_dir>/snapshots/<snapshot> in include search path

davidp135, 2021-08-17:
Hi,
Yes, I have it in work/snapshots/default/common_defines.vh:
`define RV_FPGA_OPTIMIZE 1
on line 14.

Thanks, Il remove the ahb files.

I added that search path too, but still have the same error. 
```
set_global_assignment -name SEARCH_PATH "../../../design/include"
set_global_assignment -name SEARCH_PATH "../../../work/snapshots"
set_global_assignment -name SEARCH_PATH "../../../work/snapshots/default"
```

agrobman, 2021-08-17:
github corrupted my message

you should have <work_dir>/snapshots/default in include files search list   ( directory, where swerv,config creates it's files)

param file el2_param.sv is included via `include " ..." statements

davidp135, 2021-08-17:
I think it is already where I write
`set_global_assignment -name SEARCH_PATH "../../../work/snapshots/default"`

Or am I misunderstanding? 

I assume you mean el2_param.vh rather than el2_param.sv?

agrobman, 2021-08-17:
I'm confused : what error do you have now?

agrobman, 2021-08-17:
yes *.vh file

agrobman, 2021-08-17:
el2_param_t is defined in the el2_pdef.vh generated file ..

davidp135, 2021-08-17:
Still the same error as at the start:
`Error(13406): Verilog HDL error at el2_param.vh(1): object "el2_param_t" is not declared`

I think when Quartus evaluates el2_param.vh it does not know anything about the typedef in el2_pdef.vh.

davidp135, 2021-08-17:
One change I have made is that I added the following line to the quartus project creation script:
`set_global_assignment -name VERILOG_INPUT_VERSION SYSTEMVERILOG_2012`

This should hopefully force it to treat everything as SystemVerilog, even when the file extension indicates that it is Verilog. I still have the same error though.

agrobman, 2021-08-17:
the tool should compile el2_pdef.vh before all other design files ...

davidp135, 2021-08-17:
Yes, I have been trying to get Quartus to do that and I am coming to the conclusion that it cannot. I have raised a support call with Intel but they have not come back with anything useful.

For Vivado in tools/vivado.tcl, I assume 
`set_property is_global_include true  [get_files config/el2_pdef.vh]`
is doing that job. But the equivalent does not seem to be there for Intel FPGAs.

agrobman, 2021-08-17:
and if you rename it to be *.sv? All this is strange, would the tool not recognize the el2_pdef.vh as SV source, it would complain on syntax first there, but seems like it just ignores it ...
put a garbage in this file and see if the FPGA compiler a) reads it before any other files, and b) complains about the garbage ...

davidp135, 2021-08-19:
Hi, I have got the EL2 synthesizing for the Intel Cyclone 10GX. There were a few changes required to get it to build:

**1. el2_param_t error**
This is the error we have been looking at through this thread from the start:
`Error(13406): Verilog HDL error at el2_param.vh(1): object "el2_param_t" is not declared`
As agrobman suggested the el2_pdef.vh needed to be converted to el2_pdef.sv and wrapped a package construct around it:
```
package el2_pdef;
	typedef struct packed {
		bit [7:0]      BHT_ADDR_HI;
		bit [5:0]      BHT_ADDR_LO;
                ........
	} el2_param_t;
endpackage
```

Then I edited el2_param.vh to pull in the package typedef:
```
parameter el2_pdef::el2_param_t pt = '{
```
The Quartus source tcl script compiles el2_pdef.sv then adds the rest of the header files in the search path:
```
set_global_assignment -name SYSTEMVERILOG_FILE ../../../design/include/el2_def.sv
set_global_assignment -name SYSTEMVERILOG_FILE ../../../work/snapshots/default/el2_pdef.sv
set_global_assignment -name SEARCH_PATH "../../../work/snapshots/default"
set_global_assignment -name SYSTEMVERILOG_FILE  ../../../design/el2_swerv_wrapper.sv
......
```

From talking to Intel, I am not aware another way to fix this error.

**2. Defines**
Unfortunately it looks like Quartus Prime Pro requires the include file to be manually mentioned in any file using one of it's directives. This meant that I had to add the include file at the top of most of the design files. 
`include "pd_defines.vh"
https://community.intel.com/t5/Intel-Quartus-Prime-Software/when-using-quartus-18-Why-I-can-t-put-all-my-global-define/td-p/692176

**3.RAM**
Quartus Pro will not infer a RAM if it has an asynchronous read. This is the case with ``module ram_be_``depth``x``width(`` due to this line:
`if (ME && WE) begin ram_core[ADR] <= D & WEM | ~WEM & ram_core[ADR]; Q <= 'x; end`
To get around this I had to add a register to store the ram read and convert the module to take 2 clocks (pipelined) to write. I have not fully tested it, but the simulations pass:
```
`define EL2_RAM_BE(depth, width)            \
module ram_be_``depth``x``width(            \
   input logic [$clog2(depth)-1:0] ADR,     \
   input logic [(width-1):0] D, WEM,        \
   output logic [(width-1):0] Q,            \
    `EL2_LOCAL_RAM_TEST_IO                  \
);                                          \
reg [(width-1):0] ram_core [(depth-1):0];   \
reg delay_write;                            \
reg delay_wem;                              \
reg [(width-1):0] ram_store;                \
reg [(width-1):0] delay_data;               \
`ifdef GTLSIM                               \
integer i;                                  \
initial begin                               \
   for (i=0; i<depth; i=i+1)                \
     ram_core[i] = '0;                      \
end                                         \
`endif                                      \
always @(posedge CLK) begin                 \
`ifdef GTLSIM                               \
   if (ME && WE)       ram_core[ADR] <= D & WEM | ~WEM & ram_core[ADR];      \
`else                                       \
   delay_write <= (ME && WE);               \
   if (ME && WE) begin                      \
      ram_store  <= ram_core[ADR];          \
      delay_data <= D;                      \
      delay_wem  <= WEM;                    \
      Q <= 'x;                              \
   end                                      \
   if (delay_write)                         \
      ram_core[ADR] <= delay_data & delay_wem | ~delay_wem & ram_store[ADR];  \
`endif                                      \
   if (ME && ~WE) Q <= ram_core[ADR];       \
end                                         \
assign ROP = ME;                            \
                                            \
endmodule
```

**4. module \`TEC_RV_ICG name**
In beh_lib.sw, line 751 there is a module named \`TEC_RV_ICG. I don't understand what this means? I renamed the module to a standard name `module tec_rv` and the test bench still seems to pass, but I don't know what the consequences are.

From initial builds in a Cyclone 10 GX 10CX220YF672E5G the "default" build used 28,708 / 80,330 ALMs (36%) and looks to have a max clock speed of about 48MHz. Unfortunately I don't have any hardware to load it on. I have attached the project if it is of interest.
[Cores-SweRV-EL2.zip](https://github.com/chipsalliance/Cores-SweRV-EL2/files/7017192/Cores-SweRV-EL2.zip)

olofk, 2021-08-19:
Many thanks for digging into this and publishing your work, @davidp135! I've been hoping to port the [SweRVolf SoC](https://github.com/chipsalliance/Cores-SweRVolf) to Intel (and other non-Xilinx) devices for quite some time now but haven't found the time to wade through the synthesis issues. I understand that some of the required changes are quite intrusive, but since we're using [FuseSoC](https://github.com/olofk/fusesoc) to describe the core and SoC we can actually patch the core on-the-fly to add the explicit includes and such even if they are not applied in the upstream code base. Speaking of the includes I did some searching for this to see if it was possible to compile all files in one compile unit (these implicit includes are causing headaches in pretty much all the tools I have tried but I have managed to find workarounds for most of them). I did come across this forum post https://community.intel.com/t5/FPGA-SoC-And-CPLD-Boards-And/Is-there-a-way-to-include-file-with-macros-and-small-modules/td-p/650653 which seems to indicate that it should be possible. But I haven't been able to confirm it myself yet and Quartus nowadays is a disaster with the Lite, Standard and Pro versions diverging to the point that they use different parsers with different bugs which makes it near impossible to create something that reliably works for all three versions

agrobman, 2021-08-19:
David, 

Thanks for updates.

1) However, I did not suggest convert el2_pdef.vh to package, just change file extension from .vh to .sv
another try could be  to add
` `include "el2_pdef.vh" `
line to design/include/el2_def.sv  to make the generated complex parameter type (el2_param_t) be visible from el2_pkg to all modules
(they all should have import from el2_pkg::* statements).    

then it shouldn't be need to provide el2_pdef.vh  directly as compiled file.

2) I'm not sure design will work if you'll delay write to RAMs. it was built in assumption that memories reads and writes operations take one clock and can happen back to back with the same address.
you may try to -unset icache_waypack , when running swerv.config (or just patch el2_param.vh, by assigning ICACHE_WAYPACK = 0, instead of 1) ( design will use memories with no BIT enables (_BE))

3) TEC_RV_ICG - technology specific clock gating cell name in the ASIC flow. You shouldn't use any clock gating cells with FPGA flow. generated common_defines.vh file should contain it's define

davidp135, 2021-08-20:
Hi @olofk, thanks! I think the EL2 is definitely worth the effort as it looks like a really good and solid core. Quartus certainly is a bit of a nightmare when it comes to SystemVerilog. I just had a go with what that Intel link you suggested said, and I can't get it to work unfortunately. Il take a look at Fusesoc, that is definitely useful if it can add on the fly changes to change from what is on the upstream code base. I can see one compile unit working, although have no idea how to do it!

Hi @agrobman, I did try a straight .sv change too, but it didn't help. Putting junk in the file was not noticed if the file was only in the search path, but was noticed if it was added to the compilation list.

Il try 1 and 2 and get back to you.

3. Ah yes I remember now that TEC_RV_ICG was ASIC only. Originally I wrapped a `ifndef RV_FPGA_OPTIMIZE around that module. That did not help then, but now I have the common_defines being pulled into each file correctly, wrapping the module with `ifndef RV_FPGA_OPTIMIZE **does** work without having to change the name of the module. If I do not edit the file with the ifndef I still do get a compilation error.

agrobman, 2021-08-20:
one more workaround can be to replace pt.<AAA> by `RV_<AAA>  in all design files, to get rid on the pt parameter and remove `includes for the parameter.  common_defines.vh mirrors the pt parameters as `RV_* defines  ... simple sed commands should do this.

We used this multi value parameter to allow to build several cores with different characteristics into one SOC. If you use one core in your design or all the cores are the same, converting parameters to defines should work.

seems some tools don't support structured parameters...

davidp135, 2021-08-25:
Hi @agrobman 
1. Yes the el2_def.sv change works, good idea. There are now no file name changes.
2. Yes, setting ICACHE_WAYPACK = 0 works too. What effect does this have on the core functionality?

RV_<AAA> - Yes this looks to work too.

Thanks for all your help with this @agrobman it has been really helpful.

agrobman, 2021-08-25:
ICACHE_WAYPACK does not change core functionality it effects physical build and allows to use less memory arrays instances.
Useful for ASIC flow mainly.

ukekp, 2024-04-26:
> I understand that some of the required changes are quite intrusive, but since we're using [FuseSoC](
https://github.com/olofk/fusesoc)
to describe the core and SoC we can actually patch the core on-the-fly to add the explicit includes and such even if they are not applied in the upstream code base.

Hi @olofk , I'm working on a project for a new Altera Agilex device and I am using VeeRwolf in it. How exactly can I use fusesoc to patch files?