chipsalliance/Cores-VeeR-EL2 #33: SweRV EL2 test comparison with whisper ISS

HamzaShabbir517, 2021-09-05:
Hello, 
I am trying to run the test on both SweRV EL2 and whisper ISS to check the functionality of the core before and after adding some features.
I am facing an issue that the logs files i am getting from both sides are totally different. I am generating the program.hex file from the SweRV EL2 and used it to simulate on both SweRV EL2 and whisper ISS. But the log files shows that the whisper ISS used some predefined instruction which SweRV EL2 don't. 
Is there is something which i am missing. 
I am attaching the log snaps also for reference.
Whisper ISS Log File
[image]

SweRV_EL2 Log file
[image]

Program.hex file
[image]

agrobman, 2021-09-05:
your stuff shows that whisper starts from address 0x8000_00a0, verilog from 0x8000_0000 (?) 
the simulations should start from the CPU reset vector address , which is 0x8000_0000, I guess...

HamzaShabbir517, 2021-09-05:
Yes. That is the issue i am facing as if i am running the hex file so it
should start according to the hex file. But somehow it is starting from the
address 0x800000a0. And you can see in he hex file that the instruction
whisper executing in the start are not in the program hex file. So from
does those instruction are getting generated?

agrobman, 2021-09-05:
whisper is reset to reset vector("reset_vec"), defined in json file, provided in whisper command line. You better use json file from the CPU build directory - something like .... /snapshots/default/whisper.json

HamzaShabbir517, 2021-09-05:
What is the argument to provide the whisper.json file to the whisper. As i have seen the help there is no such argument to pass the json file.

agrobman, 2021-09-05:
--configfile=.<....>/snapshots/default/whisper.json

HamzaShabbir517, 2021-09-05:
thanks got it

HamzaShabbir517, 2021-09-05:
@agrobman  After following your solution the first issue of mine is solved. Now i get the both log file exactly same. But the issue is now that my program stuck in infinite loop i am adding the snaps for the reference.

C program
[image]

Program Hex generated by Swerv
[image]

Whisper ISS Log
[image]

SweRV EL2 log
[image]

agrobman, 2021-09-05:
1) you need to provide tohost address to whisper to make it stop on tohost write. (or provide end of program address)
2) the demoTB uses fixed specific address (0xD0580000) as tohost to detect end of test condition when this location is written to 0xff

Read whisper documentation to find ways to finish its simulation. it has a few ways to do so ..

BTW, Seems your executable is over optimized - compiler removes almost all code you wrote in C ... you may try to omit -On compiler option to get more code in executable ..

HamzaShabbir517, 2021-09-05:
Thanks, I'll check it out.

agrobman, 2021-09-05:
more - you may try to use ELF formatted executable with whisper, then it may be able to extract predefined labels values automatically ..

closed 2021-09-06 by HamzaShabbir517