chipsalliance/Cores-VeeR-EL2 #34: Internal timer interrupt service routine

RehanEjaz, 2021-09-15:
### Hello 
I am a final year undergraduate student having some experience in SystemVerilog. Currently working on a project which includes integration of SweRV into SoC.
I've configured internal timer interrupt in SweRV-El2 but want to know how exactly I can write an interrupt service routine for that interrupt.
With gratitude,
Rehan Ejaz

aprnath, 2021-09-16:
Hi Rehan,

In the default (non-vectored) mode, all exceptions vector to the address specified in _mtvec_ CSR. Assuming you have not written any exception handlers, you would minimally need to

1. Set up _mtvec_ CSR to point to an address that will contain the exception handlers
2. At the address specified by _mtvec_, write a basic case statement, examining the contents of _mcause_ and branching to the appropriate handler for each exception cause
3. The handlers for the internal timer interrupt should disable the timer (otherwise you will keep getting interrupted as count will continue to be greater than bound), or re-arm the timer bounds for a future interrupt (a la periodic interrupt).
4. _mret_ to back to your program

Hope this helps

RehanEjaz, 2021-09-17:
Hi Ajay,
Thanks for your help I've successfully configured ISR for internal timer.

closed 2021-09-17 by RehanEjaz