chipsalliance/Cores-VeeR-EL2 #37: understanding Branch Prediction

KinzaQamar, 2021-12-12:
Greetings!!
I am currently understanding the branch prediction mechanism of SweRV El2 and I am captured by alot of confusions. It would be appreciated if I get to know about the following questions:
1- Overveiw of branch prediction implemented in El2
2- Why we are using hashing and folding for Ghr's and PC
3- Why there are multiple GHR's
4- How many are the algorithms involved in this module and what are those ?
5- Implementation of BHT 
6- Any example you can give to explain the whole mechansim in steps
7- LRU

shylb1996, 2024-07-09:
[image]

shylb1996, 2024-07-09:
1

shylb1996, 2024-07-09:
[image]

shylb1996, 2024-07-09:
[image]
[image]
[image]

shylb1996, 2024-07-09:
[Uploading 白色星星.svg (1).zip…]()

mgielda, 2025-11-25:
Closing as stale.