chipsalliance/Cores-VeeR-EL2 #38: Machine Timer Interrupt

HamzaShabbir517, 2022-02-16:
Hello i want to know does the feature of Machine timer interrupt is enable in the new branch of SweRV El2. As previously i got a response that 
**timer_int input is not configurable and causes another interrupt than PIC interrupts. FIR feature is not activated with timer interrupt**.

olofk, 2022-02-16:
I don't know what the FIR feature is, so perhaps I misunderstand the question, but the machine timer interrupt works fine. I use it in [SweRVolf](https://github.com/chipsalliance/Cores-SweRVolf)

HamzaShabbir517, 2022-02-16:
Can you guide me how did it use it. As i was able to configure the internal timers but was unsuccessful with the Machine Timer Interrupt

aprnath, 2022-02-16:
Hamza, the machine timer interrupt is an **input pin** to the core. Your SOC needs to have a CLINT or some other timer component. If it is programmable and on your system bus, you can program it via the core using ld/st to the memory mapped IO.

FIR is a `Fast Interrupt Redirect` mechanism that is available in EL2/EH2 which allows external interrupts via the PIC to use a vector table in the DCCM directly to speed up interrupt handling.

aprnath, 2022-02-16:
Hamza, just to re-iterate on my previous comment : `timer_int` is an input to the core. Within the core, you can use it per the RISC-V standard - it is a defined architectural function, with enable/disable via MIE and a specified `mcause`. 

How this interrupt is generated and delivered to the core is not part of EH2, but usually a timer module in your SOC or a CLINT in your SOC will provide this functionality.

This is a completely different feature than the external interrupt mechanism that the EH2 PIC provides, as indicated by @agrobman.

HamzaShabbir517, 2022-02-16:
yes i have a timer peripheral which generates the interrupt externally. But there is no information on how to configure this timer_int of the core. Can u help me with that

HamzaShabbir517, 2022-02-16:
ok thanks

closed 2022-02-16 by HamzaShabbir517