chipsalliance/Cores-VeeR-EL2 #39: display statement in el2_ifu.sv

KinzaQamar, 2022-03-25:
Hi,
I saw these lines in el2_ifu.sv file
https://github.com/chipsalliance/Cores-SweRV-EL2/blob/7045b803cab825bc3bb3dbca0cb019e55098acc4/design/ifu/el2_ifu.sv#L340-L362
If we see line num 340, it says:
https://github.com/chipsalliance/Cores-SweRV-EL2/blob/7045b803cab825bc3bb3dbca0cb019e55098acc4/design/ifu/el2_ifu.sv#L340-L341
That on every falling edge of clock, if mcycle value = 32'h00000010 then we will see all the $display statements.
However, when I write the csr mcycle with 32`h00000010 it does nothing.

Can anybody help me out about how can I generate all the display statements? As I am currently understanding branch prediction of SweRV EL2, visualizing the results with the help of display statement would speedup my progress.

Thankyou,

KinzaQamar, 2022-03-25:
[image]
I am writing 32'h00000010 on 0xB80 (address of mycylel). Am I following the correct direction?

agrobman, 2022-03-25:
it's designer's leftover to debug his stuff, I'm not sure about mcycle reference there , but you need to provide DUMP_BTB_ON define to compile this code...

KinzaQamar, 2022-03-25:
Yes, I had noticed that later. Can you let me know about how I can set this define ``DUMP_BTB_ON`` . I can't find this define other than ``el2_ifu.sv``.

agrobman, 2022-03-25:
need to `define ... in the el2_ifu.sv file .  BTW, you don't need to write to mcycle in your test.

KinzaQamar, 2022-03-26:
I have tried both ways but it didn't worked