chipsalliance/Cores-VeeR-EL2 #42: srai srli opcode swap in SweRV

RehanEjaz, 2022-04-20:
I was running some tests on SweRV and Whisper and compared them I found these logs
---
### log generated by SweRV-EL2
``` //   Cycle : #inst    0    pc    opcode    reg=value   ; mnemonic
        16 :       #1 0 80000000 595552b7  t0=59555000 ; lui     t0,0x59555000
        22 :       #2 0 80000004 55528293  t0=59555555 ; addi    t0,t0,1365
        28 :       #3 0 80000008 7c029073              ; csrrw   zero,csr_7c0,t0
        33 :       #4 0 8000000c 70041117  sp=f004100c ; auipc   sp,0x70041000
        38 :       #5 0 80000010 ff410113  sp=f0041000 ; addi    sp,sp,-12
        39 :       #6 0 80000014 00002805  ra=80000016 ; c.jal   0x80000044
        48 :       #7 0 80000044 00001141  sp=f0040ff0 ; c.addi  sp,-16
        49 :       #8 0 80000046 0000c622              ; c.swsp  s0,0xc [f0040ffc]
        50 :       #9 0 80000048 00000800  s0=f0041000 ; c.addi4spn s0,0x10
        51 :      #10 0 8000004a 41bd5593  a1=00000000 ; srli    a1,s10,27
        52 :      #11 0 8000004e 00004781  a5=00000000 ; c.li    a5,0
        53 :      #12 0 80000050 0000853e  a0=00000000 ; c.mv    a0,a5
        54 :      #13 0 80000052 00004432  s0=00000000 ; c.lwsp  s0,0xc [f0040ffc]
        55 :      #14 0 80000054 00000141  sp=f0041000 ; c.addi  sp,16
        56 :      #15 0 80000056 00008082              ; c.jr    ra
        58 :      #16 0 80000016 50580297  t0=d0580016 ; auipc   t0,0x50580000
        59 :      #17 0 8000001a fea28293  t0=d0580000 ; addi    t0,t0,-22
        60 :      #18 0 8000001e 0ff00313  t1=000000ff ; addi    t1,zero,255
        61 :      #19 0 80000022 00628023              ; sb      t1,0(t0) [d0580000]
        62 :      #20 0 80000026 00004305  t1=00000001 ; c.li    t1,1 
```
---
### log generated by Whisper
```
#1 0 80000000 595552b7 r 05         59555000  lui      x5, 0x59555
#2 0 80000004 55528293 r 05         59555555  addi     x5, x5, 1365
#3 0 80000008 7c029073 c 07c0       59555555  csrrw    x0, mrac, x5
#4 0 8000000c 70041117 r 02         f004100c  auipc    x2, 0x70041
#5 0 80000010 ff410113 r 02         f0041000  addi     x2, x2, -12
#6 0 80000014     2805 r 01         80000016  c.jal    . + 0x30
#7 0 80000044     1141 r 02         f0040ff0  c.addi   x2, -0x10
#8 0 80000046     c622 m f0040ffc   00000000  c.swsp   x8, 0xc
#9 0 80000048     0800 r 08         f0041000  c.addi4spn  x8, 0x4
#10 0 8000004a 41bd5593 r 0b         00000000  srai     x11, x26, 27
#11 0 8000004e     4781 r 0f         00000000  c.li     x15, 0x0
#12 0 80000050     853e r 0a         00000000  c.mv     x10, x15
#13 0 80000052     4432 r 08         00000000  c.lwsp   x8, 0xc
#14 0 80000054     0141 r 02         f0041000  c.addi   x2, 0x10
#15 0 80000056     8082 r 00         00000000  c.jr     x1
#16 0 80000016 50580297 r 05         d0580016  auipc    x5, 0x50580
#17 0 8000001a fea28293 r 05         d0580000  addi     x5, x5, -22
#18 0 8000001e 0ff00313 r 06         000000ff  addi     x6, x0, 255
#19 0 80000022 00628023 m d0580000   000000ff  sb       x6, 0x0(x5)
```
---

at instruction #10 there is a mismatch
```
 51 :      #10 0 8000004a 41bd5593  a1=00000000 ; srli    a1,s10,27 // EL2
```
```
#10 0 8000004a 41bd5593 r 0b         00000000  srai     x11, x26, 27 //Whisper
```

RehanEjaz, 2022-04-20:
I explored it and found that in dasm.svi
at line #308 srai and srli were swapped.
```
 5: mn = opcode[30] ? "srli": "srai"; // before
```
 I swapped them and now logs are generated as expected by whisper
```
   5: mn = opcode[30] ? "srai": "srli"; // after
```

weiyan-zhang, 2022-08-31:
I compared the exec.log with trace from other simulator. I think you are right and it should be swapped.

tmichalak, 2023-04-14:
Fixed with #77. Closing.

closed 2023-04-14 by tmichalak