chipsalliance/Cores-VeeR-EL2 #46: Synthesis dc_shell giving error while reading: parameter el2_param_t pt = '{

rahulsharmab, 2022-07-07:
Hi,

I am trying to synthesize the code for EL2 and when i try to read in the file "el2_param.vh" as an include file,
i get error at the first line as the compiler is not able to understand the code for the parameter as below.

parameter el2_param_t pt = '{

It gives the following error :
  7 Error:  /ic/dump/rahul.sharma/RISC-V/Cores-SweRV-EL2/WORK/snapshots/default/el2_param.vh:1: Syntax error at     or near token 'el2_param_t': expecting default value in local parameter. (VER-294)
  8 Error:  /ic/dump/rahul.sharma/RISC-V/Cores-SweRV-EL2/WORK/snapshots/default/el2_param.vh:1: Syntax error at     or near token 'pt'. (VER-294)

I would appreciate if you could help me with the same and guide me as to how i can synthesize this code.

Thanks,
Rahul.

[el2_param.txt](https://github.com/chipsalliance/Cores-SweRV-EL2/files/9061233/el2_param.txt)

rahulsharmab, 2022-07-07:
The dc shell version i am using is a below:
S-2021.06-SP3

Just wanted to add the information as i forgot to add it earlier.

rahuljainNV, 2023-01-18:
pls check the file order - this file should not be read directly, but only as include in the module header.

calebofearth, 2023-02-27:
Has this been resolved?
I think the problem here is that el2_pdef defines the struct used in el2_param.vh, and el2_pdef gets imported into el2_pkg, but not all files that include el2_param.vh are importing el2_pkg.
For example, see [el2_dma_ctrl.sv](https://github.com/chipsalliance/Cores-VeeR-EL2/blob/main/design/el2_dma_ctrl.sv).
I have noticed that this is not an issue when compiling with VCS (in our local flow) because the typedef gets thrown into the global namespace, but this is probably a difference in DC.

howardtr, 2023-03-23:
I was able to get this to work on my end by adding `include "el2_pdef.vh" to include/el2_dev.sv (el2_pkg), since that is getting imported right before the include. Note that the import needs to be propagated to a few files as well as @calebofearth mentioned (dec/el2_dec_tlu_ctl.sv, el2_dma_ctrl.sv, el2_pic_ctrl.sv, )

See: https://github.com/chipsalliance/caliptra-rtl/pull/35

mkurc-ant, 2023-05-29:
@calebofearth @howardtr I pulled the changes from Caliptra. Please take a look: https://github.com/chipsalliance/Cores-VeeR-EL2/pull/85

howardtr, 2023-05-29:
Thank you!  lgtm

tmichalak, 2023-06-01:
Fixed by https://github.com/chipsalliance/Cores-VeeR-EL2/pull/85