chipsalliance/Cores-VeeR-EL2 #50: Replacing hand instantiated clock gate cells

rahuljainNV, 2023-01-18:
Code has hand-instantiated clock gating cells which are defined in below library.

swerv_el2/rtl/lib/beh_lib.sv
module `TEC_RV_ICG

Generally such clock gates are not synthesized, rather a specific specially design cells is picked up from tech library as we don't want any random unbalanced combo logic on clock path, and timing has to be carefully done.

I do see this particular module being different from the rest in library and using a define for module name. So I am guessing that there was an intent to handle this differently and perhaps allow it to be replaced without having to edit the  riscv core code.

But I am not able to see how to achieve that - could someone please clarify?

nstewart-amd, 2023-02-07:
I recommend either:
a) Omit the module definition
or
b) ifdef synthesizable portion and leave intentional placeholder for per-integrator replacement.

a) Omit : force user substitution by TECH_SPECIFIC_EC_RV_ICG
751 `ifndef TECH_SPECIFIC_EC_RV_ICG
752 module `TEC_RV_ICG
...
775 endmodule
776 `endif

b) ifdef synthesizable portion and leave intentional placeholder for per-integrator replacement.
757 `ifndef TECH_SPECIFIC_EC_RV_ICG
758    logic  en_ff;
759    logic  enable;
...
773    assign Q = CK & en_ff;
774 `else  //TECH_SPECIFIC_EC_RV_ICG
775   user_tech_icg user_tech_icg_inst (.Q(Q), .SE(SE), .EN(EN), .CK(CK));  //user tech cell replacement
776 `endif //TECH_SPECIFIC_EC_RV_ICG

rahuljainNV, 2023-02-15:
looks good to me, assuming module user_clock_gate sits in area outside of the core (which it does seem so).

algrobman, 2023-02-15:
just add +define+TECH_RV_ICG=<whatever your target technology clock gate cell name> to synthesis or compilation command

Or patch this define in generated  common_defines.h

mkurc-ant, 2023-02-20:
@algrobman Adding the define is not enough as the module in question is defined inside `beh_lib.sv`: https://github.com/chipsalliance/Cores-VeeR-EL2/blob/ff9ad9bdbd88aa6d0314d2a1d921122c34c182b8/design/lib/beh_lib.sv#L752-L775

There will be name conflict with the custom one from the external tech library.

varuns-nvidia, 2023-06-28:
@rahuljainNV could you close if completed?

rahuljainNV, 2023-06-30:
@nstewart-amd  - are you able to use this option as part of Caliptra codebase?
just checking if this is working end to end with caliptra flow as well.

nstewart-amd, 2023-06-30:
> @nstewart-amd - are you able to use this option as part of Caliptra codebase? just checking if this is working end to end with caliptra flow as well.

@rahuljainNV - I think this will work.  So far, it looks like we've been picking up the default inferred gaters since the code is written to default in that direction.  My preference is that the integrator should have a "force function", that requires them to define the cell (else FAIL to compile).  It's a bit of a silent failure at compile time here and will only get picked up by post compile design checks.

Also... please note that the same scenario exists the synchronizers.  Most integrators will have a foundry/technology specific synchronizer, likely exceeding 2 flop depth.