chipsalliance/Cores-VeeR-EL2 #51: synchronizers implemented as synthesizable flops

rahuljainNV, 2023-01-18:
Synchronizers need special handling and are usually hardened cells that include both the flops and have much better MTBF.
Using dedicated cells also helps in keeping the data delay between the two flops to minimum that further helps in improving MTBF.

Its a generally accepted practice for reusable IPs to use a dedicated module for synchronizers that can be replaced by integrators with their custom cells.

VeeR core has such synchronizers at below places, I would suggest replacing them with a module which is defined at a place that integrators can replace with their own implementations.

dmi_jtag_to_core_sync.v and (el2_dbg.sv, line 277) has behavioral code for flops acting as synchronizers.

algrobman, 2023-02-15:
there is a simple way to replace synchronizers with special cells:

you can redefine rvsyncss module to instantiate specialized synchronizer cells in design/libs/beh_lib.sv or have it's description in separate module and include the file in flist file for synthesis and simulation.  

Yes, dbg module should be rewritten to use the rvsyncss with parameter of 2.

nstewart-amd, 2023-04-28:
Further to this topic; 2 FF synchronizers are not considered compliant for some teams and foundry technologies.

Our default policy/methodology defines 2 DFF synchronization for a single bit signal will be reported as violation. 
cdc report scheme two_dff -severity violation

Accordingly, dmi_jtag_to_core_sync.v is considered a non-compliant synchronizer.

mkurc-ant, 2023-05-29:
Hi. Please take a look at https://github.com/chipsalliance/Cores-VeeR-EL2/pull/56. It allows substituting a custom technology-specific cells for synchronizers.