chipsalliance/Cores-VeeR-EL2 #58: RV_ASSERT_ON is always enabled

nstewart-amd, 2023-02-17:
RISCV core has assertions guarded by `ifdef RV_ASSERT_ON​

RV_ASSERT_ON is always enabled in src/rtl/riscv_core/rtl/common_defines.sv​

`ifndef VERILATOR​
    `define RV_ASSERT_ON ​
`endif​

Recommendation:​
Add `ifndef VEEREL2_SYNTHESIS or similar in addition to VERILATOR blocker.

Additionally -
Recommend all macro defines have VEEREL2 or similar prefix.  

Example: VERILATOR -> VEEREL2_VERILATOR

Note that macro defines are global namespace.  We must avoid simple names with reasonable opportunity for collision at integration level.

algrobman, 2023-02-17:
VERILATOR is a simulator automatic define, predefined by Verilator

nstewart-amd, 2023-02-17:
The following are generic and subject to contention:

common_defines.vh
`define REGWIDTH 32
`define CLOCK_PERIOD 100
`define TOP tb_top

el2_param.vh
Most everything.
From a generic perspective, VeeR users should be able to integrate mutliple Veer configurations in parallel without collision.
Hence, anything "defined" in {veer.config -default} must not collide with defines in {veer.config -high_perf}.
Defines must be uniquified allowing multiple concurrent instantiation of differently configured VeeR cores.

nstewart-amd, 2023-02-17:
I've opened a unique case to discuss defines uniquification:
https://github.com/chipsalliance/Cores-VeeR-EL2/issues/59

This thread should continue to focus specifically on the the issue of RV_ASSERT_ON always on.

rahuljainNV, 2023-02-20:
this is dupe of https://github.com/chipsalliance/Cores-VeeR-EL2/issues/49 ?

mkurc-ant, 2023-05-29:
@nstewart-amd Please take a look at this PR https://github.com/chipsalliance/Cores-VeeR-EL2/pull/55. Does it solve the issue?