chipsalliance/Cores-VeeR-EL2 #88: Next PC not sampled when core wakes up from sleep in Verilator run

upadhyayulakiran, 2023-06-07:
Scenario:
1. Core is halted which turns off clk
2. Core wakes up (i_cpu_run_req_d1 is asserted) and tlu_flush_path_r has next PC value
3. Clk is active 1 cycle after cpu_run_req_d1 is asserted
4. tlu_flush_path_r_d1 flop does not sample the next PC value on the first clk edge after core wakes up and remains at 0
5. The 0 value is propagated to mepc, core executes from instr 0 once again and the test goes in an infinite loop

[image]

bharatpillilli, 2023-06-19:
@mgielda @kgugala - do either of you know if there is someone that owns/changes verilator code for issues like this?

tmichalak, 2023-06-21:
@upadhyayulakiran can you point me to the code with the smoke_test_clk_gating test you are referring to?

upadhyayulakiran, 2023-07-06:
@tmichalak, apologies for the late response. It looks like that test did not get merged yet. Same issue exists with this test:
src/integration/test_suites/smoke_test_cg_wdt

The test hangs and PC keeps going in a loop after core wakes up from halt.

jlmahowa-amd, 2024-01-09:
When VEER is used in the Caliptra FPGA, we observe the same behavior. I cannot build the default TEC_RV_ICG defined in beh_lib.sv for FPGA and the USER_EC_RV_ICG I can get to work isn't identical logically. I am temporarily working around this by using free_l2clk as the clock signal for flush_lower_ff.