chipsalliance/Cores-VeeR-EL2 #107: Issue with dhry test

cartersmith15, 2023-08-21:
I'm trying to run the dhry test using this command:   "make -f $RV_ROOT/tools/Makefile verilator TEST=dhry"

I am met with this error:

 /home/csmith/Cores-VeeR-EL2/third_party/picolibc/install/picolibc/riscv64-unknown-elf/include/inttypes.h:313:3: error: unknown type name 'intmax_t'
  313 |   intmax_t      quot;
/home/csmith/Cores-VeeR-EL2/third_party/picolibc/install/picolibc/riscv64-unknown-elf/include/inttypes.h:324:8: error: unknown type name 'uintmax_t'
  324 | extern uintmax_t strtoumax(const char *__restrict, char **__restrict, int);
/home/csmith/Cores-VeeR-EL2/third_party/picolibc/install/picolibc/riscv64-unknown-elf/include/stdio.h:81:9: error: unknown type name 'uint16_t'
   81 | typedef uint16_t __ungetc_t;
      |         ^~~~~~~~
/home/csmith/Cores-VeeR-EL2/third_party/picolibc/install/picolibc/riscv64-unknown-elf/include/stdio.h:308:1: error: unknown type name 'ssize_t'; did you mean '_ssize_t'?
  308 | ssize_t getdelim(char **__restrict lineptr, size_t *__restrict  n, int delim, FILE *__restrict stream);
      | ^~~~~~~

I don't understand why it is not working. I was able to get the hello_world function to work but this one is not. 

Thanks for any help. :)

mczyz-antmicro, 2023-08-30:
Hey @cartersmith15,

the test you mention works in [CI](https://github.com/chipsalliance/Cores-VeeR-EL2/actions/runs/6023092653/job/16339153452#step:8:2027) (Ubuntu 22.04) and in my local environment (Debian 11), so I was unable to reproduce your results. Which OS are you using? I believe that the problem must be in your environment.

You can try retracing steps from CI, which are:
1. [Build Verilator](https://github.com/chipsalliance/Cores-VeeR-EL2/blob/main/.github/workflows/build-verilator.yml)
   1. Make sure you've installed these [packages](https://github.com/chipsalliance/Cores-VeeR-EL2/blob/main/.github/workflows/build-verilator.yml#L37C37-L37C37)
2. [Test Regression](https://github.com/chipsalliance/Cores-VeeR-EL2/blob/main/.github/workflows/test-regression.yml#L23)
   1. Make sure you've installed these [packages](https://github.com/chipsalliance/Cores-VeeR-EL2/blob/main/.github/workflows/test-regression.yml#L23)
   2. Final environment [setup](https://github.com/chipsalliance/Cores-VeeR-EL2/blob/main/.github/workflows/test-regression.yml#L71-L85)

I would also suggesting checking if you checked out picolibc correctly (compare commit SHA) and/or how is riscv*-unknown-elf installed in your environment?

tmichalak, 2024-09-11:
@cartersmith15 I belive you were able to resolve the problem using the abovementioned steps. Closing this for now, but feel free to comment if your problem perisists.

closed 2024-09-11 by tmichalak