chipsalliance/Cores-VeeR-EL2 #133: Does not build

algrobman, 2023-11-19:
I'm running :  
make -f $RV_ROOT/tools/Makefile

rm -rf *.log *.s *.hex *.dis *.tbl irun* vcs* simv* *.map snapshots veer* \
verilator* *.exe obj* *.o *.sym ucli.key vc_hdrs.h csrc *.csv work \
dataset.asdb library.cfg vsimsa.cfg riviera-build wave.asdb
make -f el2/Cores-VeeR-EL2/tools/picolibc.mk all
make[1]: Entering directory `el2'
cd el2/Cores-VeeR-EL2/third_party/picolibc && meson el2/Cores-VeeR-EL2/third_party/picolibc/build \
	-Dmultilib=true \
	-Dmultilib-list=rv32imac/ilp32 \
	-Dpicocrt=false \
	-Datomic-ungetc=false \
	-Dthread-local-storage=false \
	-Dio-long-long=true \
	-Dformat-default=integer \
	-Dincludedir=picolibc/riscv64-unknown-elf/include \
	-Dlibdir=picolibc/riscv64-unknown-elf/lib \
        -Dprefix=el2/Cores-VeeR-EL2/third_party/picolibc/install \
        -Dspecsdir=el2/Cores-VeeR-EL2/third_party/picolibc/install \
	--cross-file el2/Cores-VeeR-EL2/third_party/picolibc/build/cross.txt

ERROR: Neither directory contains a build file meson.build.
make[1]: *** [el2/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs] Error 1
make[1]: Leaving directory `el2'
make: *** [picolibc] Error 2

what's wrong?

robertszczepanski, 2023-11-20:
Hi @algrobman, it looks like the error you've attached occurs when you try to build VeeR core without having all required submodules.

Please make sure that command `git submodule update --init --recursive` executes without errors. Since you've already run build command you might have to first run `git submodule deinit --force .` (note the dot at the end).

algrobman, 2023-11-20:
BTW, what is this picolibs stuff for?

algrobman, 2023-11-20:
Still see picolibc build failures:

cd /fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc && meson /fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc/build \
	-Dmultilib=true \
	-Dmultilib-list=rv32imac/ilp32 \
	-Dpicocrt=false \
	-Datomic-ungetc=false \
	-Dthread-local-storage=false \
	-Dio-long-long=true \
	-Dformat-default=integer \
	-Dincludedir=picolibc/riscv64-unknown-elf/include \
	-Dlibdir=picolibc/riscv64-unknown-elf/lib \
        -Dprefix=/fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc/install \
        -Dspecsdir=/fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc/install \
	--cross-file /fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc/build/cross.txt
DEPRECATION: c_args in the [properties] section of the machine file is deprecated, use the [built-in options] section.
The Meson build system
Version: 0.59.0
Source dir: /fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc
Build dir: /fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc/build
Build type: cross build
Project name: picolibc
Project version: 1.8
C compiler for the host machine: riscv64-unknown-elf-gcc (gcc 10.1.0 "riscv64-unknown-elf-gcc (GCC) 10.1.0")
C linker for the host machine: riscv64-unknown-elf-gcc ld.bfd 2.35
C compiler for the build machine: cc (gcc 4.8.5 "cc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)")
C linker for the build machine: cc ld.bfd 2.34
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: riscv64
Host machine cpu: riscv
Target machine cpu family: riscv64
Target machine cpu: riscv
Compiler for C supports arguments -nostdlib: YES 
Checking if "long double check" compiles: YES 
Fetching value of define "__SIZEOF_FLOAT__" : 4 
Fetching value of define "__SIZEOF_DOUBLE__" : 8 
Fetching value of define "__SIZEOF_LONG_DOUBLE__" : 16 
Compiler for C supports arguments -fno-common: YES 
Compiler for C supports arguments -frounding-math: YES 
Compiler for C supports arguments -fsignaling-nans: YES 
Compiler for C supports arguments -Wno-unsupported-floating-point-opt: NO 
Compiler for C supports arguments -fno-stack-protector: YES 
Compiler for C supports arguments -fno-builtin-copysignl: YES 
Program riscv64-unknown-elf-gcc-nm found: YES
Program scripts/duplicate-names found: YES (/fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc/scripts/duplicate-names)
Compiler for C supports link arguments -Wl,--defsym=_start=0: YES 
Compiler for C supports link arguments -Wl,-alias,main,testalias: NO 
Compiler for C supports function attribute alias: YES 
Compiler for C supports function attribute format: YES 
Compiler for C supports function attribute weak: YES 
Configuring picolibc.specs using configuration
Configuring picolibcpp.specs using configuration
Configuring picolibc.ld using configuration
Configuring picolibcpp.ld using configuration
Compiler for C supports arguments -Werror=implicit-function-declaration: YES 
Compiler for C supports arguments -Werror=vla: YES 
Compiler for C supports arguments -Warray-bounds: YES 
Compiler for C supports arguments -Wold-style-definition: YES 
Compiler for C supports arguments -Werror=double-promotion: YES 
Compiler for C supports arguments -Wno-missing-braces: YES 
Compiler for C supports arguments -Wno-implicit-int: YES 
Compiler for C supports arguments -Wno-return-type: YES 
Compiler for C supports arguments -Wno-unused-command-line-argument: NO 
Checking if "packed structs may contain bitfields" compiles: YES 
Checking if "has __builtin_mul_overflow" links: YES 
Checking if "has __builtin_add_overflow" links: YES 
Checking if "supports _Complex" compiles: YES 
Checking if "has __builtin_expect" links: YES 
Compiler for C supports arguments -Werror: YES 
Checking if "attribute __alloc_size__" compiles: YES 
Checking if "attributes constructor/destructor" compiles: YES 
Message: target .;

meson.build:920:1: ERROR: Problem encountered: Unavailable multilib: rv32imac/ilp32

A full log can be found at /fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc/build/meson-logs/meson-log.txt
make[1]: *** [/fac/proj_wy/users/grobman/work/S200/el2/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs] Error 1
make[1]: Leaving directory `/fac/proj_wy/users/grobman/work/S200/el2'
make: *** [picolibc] Error 2

tmichalak, 2023-11-20:
This error means that the toolchain you're using doesn't have support for the 32-bit RISC-V architecture with the specified ISA extensions and ABI. You need a toolchain with multilib support (see #132).

tmichalak, 2023-11-20:
> BTW, what is this picolibs stuff for?

picolibc is a fully BSD licensed revised version of newlibc which uses a lightweight stdio lib, which is more suitable for low memory devices which VeeR is meant for.

algrobman, 2023-11-20:
Don't you think that not all users need this picolibc and it shouldn't be built in main Makefile if I want to run just hello_world?

closed 2024-03-08 by algrobman