chipsalliance/Cores-VeeR-EL2 #134: Compilation fails

algrobman, 2023-11-19:
README says that I can use verilator 4.102 and higher:
I'm using verilator 4.210  and getting following:

make -f $RV_ROOT/tools/Makefile verilator-build

%Error: Unknown warning specified: -Wno-IMPLICITSTATIC... Suggested alternative: '-Wno-IMPLICIT'
make: *** [verilator-build] Error 1

mkurc-ant, 2023-11-20:
Hi. Your version is supported indeed but requires a different commandline flag(s). I updated the Makefile to detect version automatically and use correct flags in each case. Please take a look: https://github.com/chipsalliance/Cores-VeeR-EL2/pull/135

algrobman, 2023-11-20:
How do verilator version earlier than 5.000 work with latest UVM TB additions to this data base?

mkurc-ant, 2023-11-20:
UVM-based tests currently use Verilator from a branch which hasn't been merged upstream yet. See https://github.com/antmicro/Cores-VeeR-EL2/blob/mkurc/gh-133/.github/workflows/build-verilator.yml#L16-L18

closed 2024-03-08 by algrobman