chipsalliance/Cores-VeeR-EL2 #136: compile fail with TEST=dhry

hbhbts, 2023-11-21:
I have tried hello_world and core mark testcase, they can pass but for dhry it's falied.

CMD: make -f tools/Makefile TEST=dhry

the log as follow:
scv64-unknown-elf-cpp -Isnapshots/default  /home/hht/work/Cores-VeeR-EL2/testbench/tests/dhry/crt0.s  > crt0.cpp.s
riscv64-unknown-elf-as -mabi=ilp32 -march=rv32imac_zicsr crt0.cpp.s -o crt0.o
riscv64-unknown-elf-gcc -Isnapshots/default --specs=/home/hht/work/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs -g -O3 -DCOMPILER_FLAGS="\"-g -O3\"" -mabi=ilp32 -march=rv32imac_zicsr -c /home/hht/work/Cores-VeeR-EL2/testbench/tests/dhry/dhry_1.c -o dhry_1.o
riscv64-unknown-elf-gcc -Isnapshots/default --specs=/home/hht/work/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs -g -O3 -DCOMPILER_FLAGS="\"-g -O3\"" -mabi=ilp32 -march=rv32imac_zicsr -c /home/hht/work/Cores-VeeR-EL2/testbench/tests/dhry/dhry_2.c -o dhry_2.o
riscv64-unknown-elf-gcc -Isnapshots/default --specs=/home/hht/work/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs -g -O3 -DCOMPILER_FLAGS="\"-g -O3\"" -mabi=ilp32 -march=rv32imac_zicsr -c /home/hht/work/Cores-VeeR-EL2/testbench/tests/dhry/printf.c -o printf.o
Building dhry
riscv64-unknown-elf-gcc -mabi=ilp32 -march=rv32imac_zicsr --verbose -Wl,-Map=dhry.map -Tsnapshots/default/link.ld --specs=/home/hht/work/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs  -nostartfiles crt0.o dhry_1.o dhry_2.o printf.o -o dhry.exe
Using built-in specs.
Reading specs from /home/hht/work/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs
rename spec link to picolibc_link
rename spec cpp to picolibc_cpp
rename spec cc1 to picolibc_cc1
rename spec cc1plus to picolibc_cc1plus
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/hht/software/riscv64-multilib-elf/bin/../libexec/gcc/riscv64-unknown-elf/12.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /root/work/riscv-gnu-toolchain/build/../gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv64-multilib-elf --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g2ee5e430018 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv64-multilib-elf/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../../gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (g2ee5e430018) 
COMPILER_PATH=/home/hht/software/riscv64-multilib-elf/bin/../libexec/gcc/riscv64-unknown-elf/12.2.0/:/home/hht/software/riscv64-multilib-elf/bin/../libexec/gcc/:/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/:/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/:/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/lib/:/home/hht/software/riscv64-multilib-elf/bin/../riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-mabi=ilp32' '-march=rv32imac_zicsr' '-v' '-T' 'snapshots/default/link.ld' '-specs=/home/hht/work/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs' '-nostartfiles' '-o' 'dhry.exe' '-mtune=rocket' '-misa-spec=20191213' '-march=rv32imac_zicsr' '-dumpdir' 'dhry.'
 /home/hht/software/riscv64-multilib-elf/bin/../libexec/gcc/riscv64-unknown-elf/12.2.0/collect2 -plugin /home/hht/software/riscv64-multilib-elf/bin/../libexec/gcc/riscv64-unknown-elf/12.2.0/liblto_plugin.so -plugin-opt=/home/hht/software/riscv64-multilib-elf/bin/../libexec/gcc/riscv64-unknown-elf/12.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccdLyDzZ.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -L/home/hht/work/Cores-VeeR-EL2/third_party/picolibc/install/picolibc/riscv64-unknown-elf/lib/. -L/home/hht/work/Cores-VeeR-EL2/third_party/picolibc/install/picolibc/riscv64-unknown-elf/lib --sysroot=/home/hht/software/riscv64-multilib-elf/bin/../riscv64-unknown-elf -melf32lriscv --gc-sections -o dhry.exe -L/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0 -L/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc -L/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/lib -L/home/hht/software/riscv64-multilib-elf/bin/../riscv64-unknown-elf/lib -Map=dhry.map crt0.o dhry_1.o dhry_2.o printf.o -lgcc --start-group -lgcc -lc --end-group -lgcc -T snapshots/default/link.ld
/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-strcmp.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /home/hht/software/riscv64-multilib-elf/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-strcmp.o)
collect2: error: ld returned 1 exit status
make: *** [tools/Makefile:162: program.hex] Error 1

koluckirafal, 2023-11-22:
PR #139 should allow you to build tests with toolchains based on GCC 12.1 or newer, could you test it locally with your toolchain?

tmichalak, 2024-09-11:
@hbhbts closing the issue for now since there has been no answer for some time. Feel free to comment if the issue persists.

closed 2024-09-11 by tmichalak