chipsalliance/Cores-VeeR-EL2 #137: Swerv RISC V

navroops17, 2023-11-22:
Hi 
I am trying to synthesize Swerv a RISC V and can anyone help me out how can I do it. Where to do and how to do.
Please

mkurc-ant, 2023-11-24:
Hi.

The first thing to do is configuring VeeR. Please refer to the README.md (this point https://github.com/chipsalliance/Cores-VeeR-EL2#building-a-model) how to do that.

Following steps depend on the synthesis toolchain you use. The `flist` file (https://github.com/chipsalliance/Cores-VeeR-EL2/blob/main/design/flist) lists all VeeR sources needed for synthesis / simulation. You also need to add include path that points to the generated configuration (by default it generates under `snapshots/default`).

mgielda, 2024-02-01:
Closing, assuming that this is solved.

closed 2024-02-01 by mgielda