chipsalliance/Cores-VeeR-EL2 #145: SRAM instance export

calebofearth, 2023-12-11:
Request that SRAM instances (ICCM, DCCM, ICache) be moved outside of the VeeR core. Connections from internal modules should be exported to allow integrators to connect proprietary SRAM components. This could be done similarly to how Caliptra has implemented it with a [SystemVerilog interface](https://github.com/chipsalliance/caliptra-rtl/blob/main/src/riscv_core/veer_el2/rtl/lib/el2_mem_if.sv).

As part of this feature change, we also request that the SRAM interface be refactored to make the separation between data and ECC more clear. This request tracks https://github.com/chipsalliance/caliptra-rtl/issues/125. Currently, the internal memories have multiple banks, with data and ecc aggregated into a single data bus for each bank, which makes the integrator's job more difficult when implementing system level reliability checks.

algrobman, 2024-03-04:
there would be better way to use proprietary SRAMs RTL memories library modules can be replaced with custom onces, instantiating memory vendors verilogs ...

algrobman, 2024-03-08:
does this really work?
see https://github.com/chipsalliance/Cores-VeeR-EL2/issues/163

kiryk, 2024-07-26:
Hi, @calebofearth. I think now as we have the memories exported and ECC separated, this issue can be closed.

The problems mentioned in #163 wer addressed in #194.

closed 2024-07-26 by kiryk