chipsalliance/Cores-VeeR-EL2 #159: Caliptra JTAG TdoEn missing output

nstewart-amd, 2024-02-09:
“IEEE1149.1 standard says in chapter 4.5.1 “The TDO driver shall be set to its inactive drive state except when the scanning of data is in progress”. So TdoEn is needed to implement this requirement, detailed in Table 6.2.”

Caliptra RTL currently does not expose TdoEn as a top-level port.
However, TdoEn appears to be implemented in the VEER TAP controller.

File: riscv_core/veer_el2/rtl/dmi/rvjtag_tap.v
assign tdoEnable = shift_dr | shift_ir;

However, this output is left dangling at the VEER wrapper and not pulled to the top of Caliptra.

File: riscv_core/veer_el2/rtl/el2_veer_wrapper.sv
.tdoEnable (),

algrobman, 2024-03-04:
tdoEnable output is meant to tristate real SOC TDO pin. Since debug subsystem is not alone is a SOC, but connected to the SOC JTAG someway, there is no sense to use VEER tdoEnable in the SOC level.

nstewart-amd, 2024-03-05:
We should not assume that VeeR is not alone in the system.
VeeR tdoEnable should be output of VeeR, and system integrators should use it or ignore as appropriate for their systems.

PR #182 (2024-07-03): Sync code with caliptra-rtl repo
This PR brings partial changes from [caliptra-rtl repo](https://github.com/chipsalliance/caliptra-rtl) that have happened since the last code sync. These include:
- lint changes
- exposing jtag_tdoEn in el2_veer_wrapper (fixes #159)
closed 2024-07-03 by kgugala