chipsalliance/Cores-VeeR-EL2 #163: Does this design work with VCS/Cadence/Mentor simulators after memories were moved from design to tb_top?

algrobman, 2024-03-03:
I'm trying to simulate it with Xcellium  and getting whole design in Xs after first access to DCCM.
It seems to me that the memories moved to TB are not connected/wrongly connected to design:

        **assign el2_mem_export.dccm_wr_ecc_bank[i] = dccm_wr_fdata_bank[i][pt.DCCM_FDATA_WIDTH-1:pt.DCCM_DATA_WIDTH];
        assign dccm_bank_fdout[i] = {el2_mem_export.dccm_bank_ecc[i], el2_mem_export.dccm_bank_dout[i]};**

        if (DCCM_INDEX_DEPTH == 32768) begin : dccm
            ram_32768x39  dccm_bank (
                                    // Primary ports
                                    .ME(el2_mem_export.dccm_clken[i]),
                                    .CLK(el2_mem_export.clk),
                                    .WE(el2_mem_export.dccm_wren_bank[i]),
                                    .ADR(el2_mem_export.dccm_addr_bank[i]),
                                    .D(dccm_wr_fdata_bank[i][pt.DCCM_FDATA_WIDTH-1:0]),
                                    .Q(dccm_bank_fdout[i][pt.DCCM_FDATA_WIDTH-1:0]),
                                    .ROP ( ),
                                    // These are used by SoC
                                    `EL2_LOCAL_DCCM_RAM_TEST_PORTS
                                    .*
                                    );
        end

dccm_bank_fdout - is memory output,  but it is driven by interface signal in above assign statement  
dccm_wr_fdata_bank is the memory input, but it used to drive the el2_mem_export interface ????

More questions:
why were the memories moved from design? 
why do you keep sized memories macroes? They were used originally to include vendor's real memories of specific technology 
into RTL design for synthesis, but once they were  moved out why do you need them?

Also why do you moved out only ICCM/DCCM memories and not IC mems?

Are you ever simulated this design with commercial simulators?

Also  your picolibc stuff does not compile, even if I want to run just hello_world  ( asks for multilibs and ninja) could you guys, please, separate all these nice to have things from basic stuff,  so people will not need to install all this SW to just try this design?

kiryk, 2024-06-20:
@algrobman this issue with testbenches was likely fixed in https://github.com/chipsalliance/Cores-VeeR-EL2/pull/194. Do these errors still occur to you?

kiryk, 2024-06-20:
@algrobman this issue with testbenches was likely fixed in https://github.com/chipsalliance/Cores-VeeR-EL2/pull/194. Do you still get these errors after the change?

kiryk, 2024-07-17:
I'm closing this issue due to lack of activity, but feel free to reopen if the problem occurs again.

closed 2024-07-17 by kiryk