chipsalliance/Cores-VeeR-EL2 #164: Error while build through "make irun"

ajaysilla, 2024-03-05:
Hi

We downloaded the [Cores-VeeR-EL2] package and planning to execute a testcase as per the README file w.r.t the command "make irun".

We are facing below issue and kindly help in sharing the steps to resolve. (Running in linux environment)

ajaysilla, 2024-03-05:
@tmichalak  Below is the issue

cd /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc && meson /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/build \
	-Dmultilib=true \
	-Dmultilib-list=rv32imac/ilp32 \
	-Dpicocrt=false \
	-Datomic-ungetc=false \
	-Dthread-local-storage=false \
	-Dio-long-long=true \
	-Dformat-default=integer \
	-Dincludedir=picolibc/riscv64-unknown-elf/include \
	-Dlibdir=picolibc/riscv64-unknown-elf/lib \
        -Dprefix=/user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/install \
        -Dspecsdir=/user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/install \
	--cross-file /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/build/cross.txt
DEPRECATION: c_args in the [properties] section of the machine file is deprecated, use the [built-in options] section.
The Meson build system
Version: 1.3.2
Source dir: /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc
Build dir: /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/build
Build type: cross build
Project name: picolibc
Project version: 1.8

meson.build:35:0: ERROR: Unknown compiler(s): [['riscv64-unknown-elf-gcc']]
The following exception(s) were encountered:
Running `riscv64-unknown-elf-gcc --version` gave "[Errno 2] No such file or directory: 'riscv64-unknown-elf-gcc'"

A full log can be found at /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/build/meson-logs/meson-log.txt
WARNING: Running the setup command as `meson [options]` instead of `meson setup [options]` is ambiguous and deprecated.
make[1]: *** [/user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/install/picolibc.specs] Error 1
make[1]: Leaving directory `/user/ajay.silla/Cores-VeeR-EL2/run'
make: *** [picolibc] Error 2
[ajay.silla]: /user/ajay.silla/Cores-VeeR-EL2/run> cc -print-search-dirs
install: /usr/lib/gcc/x86_64-redhat-linux/4.8.5/
programs: =/usr/libexec/gcc/x86_64-redhat-linux/4.8.5/:/usr/libexec/gcc/x86_64-redhat-linux/4.8.5/:/usr/libexec/gcc/x86_64-redhat-linux/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/:/usr/lib/gcc/x86_64-redhat-linux/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../x86_64-redhat-linux/bin/x86_64-redhat-linux/4.8.5/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../x86_64-redhat-linux/bin/
libraries: =/usr/lib/gcc/x86_64-redhat-linux/4.8.5/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../x86_64-redhat-linux/lib/x86_64-redhat-linux/4.8.5/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../x86_64-redhat-linux/lib/../lib64/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../x86_64-redhat-linux/4.8.5/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../lib64/:/lib/x86_64-redhat-linux/4.8.5/:/lib/../lib64/:/usr/lib/x86_64-redhat-linux/4.8.5/:/usr/lib/../lib64/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../x86_64-redhat-linux/lib/:/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../:/lib/:/usr/lib/

tmichalak, 2024-03-05:
You need a toolchain for the RISC-V ISA. Not sure if redhat has a gcc-riscv64-unknown-elf package, but you can download it from [here](https://github.com/chipsalliance/caliptra-tools/releases/tag/gcc-v12.1.0)

ajaysilla, 2024-03-07:
Thanks, @tmichalak, for the quick resolution.

After using the above package , we are observing the below issue

Build started at 2024-03-07T17:44:37.184597
Main binary: /appl/CAEutil/LINUX7/local/Python-3.8.1/bin/python3
Build Options: -Dmultilib=true -Dmultilib-list=rv32imac/ilp32 -Dpicocrt=false -Datomic-ungetc=false -Dthread-local-storage=false -Dio-long-long=true -Dformat-default=integer -Dincludedir=picolibc/riscv64-unknown-elf/include -Dlibdir=picolibc/riscv64-unknown-elf/lib -Dprefix=/user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/install -Dspecsdir=/user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/install --cross-file=/user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/build/cross.txt
Python system: Linux
The Meson build system
Version: 1.3.2
Source dir: /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc
Build dir: /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/build
Build type: cross build
Project name: picolibc
Project version: 1.8
-----------
Detecting compiler via: `riscv64-unknown-elf-gcc --version` -> 0
stdout:
riscv64-unknown-elf-gcc (g1ea978e3066) 12.1.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-----------
Running command: riscv64-unknown-elf-gcc -E -dM -
-----
-----------
Detecting linker via: `riscv64-unknown-elf-gcc -Wl,--version` -> 1
stderr:
collect2 version 12.1.0
/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/bin/ld -plugin /user/ajay.silla/RISCV_TOOL/riscv/bin/../libexec/gcc/riscv64-unknown-elf/12.1.0/liblto_plugin.so -plugin-opt=/user/ajay.silla/RISCV_TOOL/riscv/bin/../libexec/gcc/riscv64-unknown-elf/12.1.0/lto-wrapper -plugin-opt=-fresolution=/tmp/cc1p6KxJ.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgloss -plugin-opt=-pass-through=-lgcc --sysroot=/user/ajay.silla/RISCV_TOOL/riscv/bin/../riscv64-unknown-elf -melf64lriscv /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/lib/crt0.o /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/crtbegin.o -L/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0 -L/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc -L/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/lib -L/user/ajay.silla/RISCV_TOOL/riscv/bin/../riscv64-unknown-elf/lib --version -lgcc --start-group -lc -lgloss --end-group -lgcc /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/crtend.o
/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/bin/ld: /lib64/libc.so.6: version `GLIBC_2.27' not found (required by /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/bin/ld)
collect2: error: ld returned 1 exit status
-----------

meson.build:35:0: ERROR: Unable to detect linker for compiler `riscv64-unknown-elf-gcc -Wl,--version`
stdout: 
stderr: collect2 version 12.1.0
/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/bin/ld -plugin /user/ajay.silla/RISCV_TOOL/riscv/bin/../libexec/gcc/riscv64-unknown-elf/12.1.0/liblto_plugin.so -plugin-opt=/user/ajay.silla/RISCV_TOOL/riscv/bin/../libexec/gcc/riscv64-unknown-elf/12.1.0/lto-wrapper -plugin-opt=-fresolution=/tmp/cc1p6KxJ.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgloss -plugin-opt=-pass-through=-lgcc --sysroot=/user/ajay.silla/RISCV_TOOL/riscv/bin/../riscv64-unknown-elf -melf64lriscv /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/lib/crt0.o /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/crtbegin.o -L/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0 -L/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc -L/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/lib -L/user/ajay.silla/RISCV_TOOL/riscv/bin/../riscv64-unknown-elf/lib --version -lgcc --start-group -lc -lgloss --end-group -lgcc /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/crtend.o
/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/bin/ld: /lib64/libc.so.6: version `GLIBC_2.27' not found (required by /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/bin/ld)
collect2: error: ld returned 1 exit status

-> Before the above error, the below error was reported and after adding the lib path to $LD_LIBRARY_PATH we are seeing the above issue
The Meson build system
Version: 1.3.2
Source dir: /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc
Build dir: /user/ajay.silla/Cores-VeeR-EL2/third_party/picolibc/build
Build type: cross build
Project name: picolibc
Project version: 1.8

meson.build:35:0: ERROR: Unable to detect GNU compiler type:
Compiler stdout:

-----
Compiler stderr:
/user/ajay.silla/RISCV_TOOL/riscv/bin/../libexec/gcc/riscv64-unknown-elf/12.1.0/cc1: error while loading shared libraries: libmpfr.so.6: cannot open shared object file: No such file or directory

Can you also please confirm, the above steps we are following are proper or not., if not let us know where we are going wrong.

Thanks,
Ajay.

MVJSK, 2024-03-14:
Hi anyone help on this? we are being stuck here since a week

robertszczepanski, 2024-03-14:
Hi,
the following error you've encountered:
```
/user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/bin/ld: /lib64/libc.so.6: version `GLIBC_2.27' not found (required by /user/ajay.silla/RISCV_TOOL/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.1.0/../../../../riscv64-unknown-elf/bin/ld)
collect2: error: ld returned 1 exit status
```
is related to the version of the C library available in your system. It means that your Linux distribution is probably too old to provide required `2.27` or higher version.

Solving this issue is possible by either upgrading your OS so it will provide more recent C library or building such library on your own. However, the latter solution might result in breaking your system since its packages might not be compatible with new version. Having said that, I would suggest upgrading your OS or moving to the already tested Linux distribution, VeeR has been tested on Debian 11 (bullseye) and 12 (bookworm) so either of these should work flawlessly for you.

tmichalak, 2024-05-16:
Closing since this is most likely an user error. @ajaysilla @MVJSK feel free to reopen and provide more details if you're still stuck.

closed 2024-05-16 by tmichalak