chipsalliance/Cores-VeeR-EL2 #250: LSU AXI ID is non-constant

calebofearth, 2024-10-22:
LSU AXI ID (ARID and AWID) is determined by a buffer pointer.
Caliptra requires the AXI ID to be a constant value, so that access privileges in the system can be maintained. 

Feature request: 
- LSU (and IFU) AXI ID should have an operating mode that allows a deterministic ID to be used. This will also require strict access ordering on the AXI interface, as per the AXI specification.
- In this operating mode, AXI ID should also be tied to the active Privilege Level (i.e. Machine Mode vs User Mode have different AXI ID).

calebofearth, 2024-10-22:
@bharatpillilli @vsonims FYI

algrobman, 2024-10-22:
AxID , RID and BID are used to relate requests and responses on AXI bus to allow out of order responses, speeding up bus operation. If you need bus with in-order response use AHB option. BTW, AxPROT fields of the   AXI provide info of what mode the processor is  in (user or privileged) if this is not sufficient, AxUSER field can be added.
Transactions on LSU bus can be serialized if used in application  memory regions are defined as "side-effect" by programming MRAC CSR

bharatpillilli, 2024-10-23:
We understand that but due to certain sic construction use models, we have made this request to allow us to have a configuration that we stated above.

calebofearth, 2024-11-12:
The architectural decision has been made to use AxUSER in Caliptra instead of AxID for agent identification, so this feature is no longer necessary. Closing.

closed 2024-11-12 by calebofearth