chipsalliance/Cores-VeeR-EL2 #258: Support PMP without requiring USER MODE

calebofearth, 2024-11-07:
Currently, when attempting to configure the VeeR core with SMEPMP=1 and USER_MODE=0, I see
`veer: ERROR! Currently Smepmp feature cannot be enabled when user mode support is disabled`
Request to support this feature.

tmichalak, 2024-11-13:
@calebofearth PMP itselft is enabled regardless of whether User mode is enabled or not. 
As for the SMEPMP and USER_MODE defines, note that SMEPMP is used to enable the [extended PMP](https://github.com/riscvarchive/riscv-tee/blob/main/Smepmp/Smepmp.pdf) which is meant for CPUs with User mode support, so you can't have one without the other defined for a given core. If you just want the PMP unit (as specified in the privileged spec) then it's enabled by default (no defines are needed).

calebofearth, 2024-11-13:
@tmichalak Thanks for the explanation, this clears it up. I missed the linked doc on [smepmp](https://github.com/riscv/riscv-tee/blob/main/Smepmp/Smepmp.pdf) on my initial read-through of the new VeeR PRM section.
I'll disable SMEPMP in our usage.

closed 2024-11-13 by calebofearth