chipsalliance/Cores-VeeR-EL2 #407: [DCLS] Evaluate physical hardening of corruption error and disable corruption detection signals.

moidx, 2025-05-21:
This issue is to document an assessment of the security claims of the shadow control and equivalence check reporting signals: 

* `disable_corruption_detection_i`
* `lockstep_err_injection_en_i`
* `corruption_detected_o`

Action items: 

* Discuss the likelihood of an attacker being able to leverage the single bit representation of the control and reporting signals to bypass the DCLS fault detection logic. 
* Consider using multi-bit encoding for control and reporting signals. Provide any additional recommendations for hardening.
* Document integration guidelines to avoid single points of failure at the SoC integration side.

johannheyszl, 2025-05-21:
`disable_corruption_detection_i` should be multi-bit encoded. Since a fault might corrupt the single FF outputting this value. 

`lockstep_err_injection_en_i` can be, but does not need to be because a fault will only lead to a corruption detection.

`corruption_detected_o` is likely asserted repeatedly in subsequent cycles upon diverging core states. It seems advisable to multi-bit encode to prevent single cycle manipulations.

amullick007, 2025-06-09:
### Proposal for `corruption_detected_o` Signal Encoding

In the current implementation of the DCLS (Deterministic Constant Latency Secure) feature, the `corruption_detected_o` signal from the DCLS comparator is a single-bit signal. We propose changing it to a 2-bit signal for better error indication and handling.

The new encoding would be as follows:
- **01**: No error detected (default state).
- **10**: Error detected
- **00** and **11**: Error states (these encodings could only be reached through single-bit signal manipulation, indicating potential corruption or tampering).

This approach ensures that any unintended or malicious signal manipulations are flagged as errors, adding an additional layer of security and integrity to the system.

Let me know your thoughts.

kgugala, 2025-08-08:
#411 introduces mubi encoding following the above specification

On Caliptra level, corruption detection bits will be routed to the `CPTRA_HW_ERROR_FATAL` and latched in a W1C register field. Least significant bit of the `corruption_detected_o` signal will be negated before it is latched. This way the False value (`2’b01`)  will be encoded in the register as `2'b00` and any other value will be considered as fault. This will let us use W1C access type (software will be able to clear any set bit).